adafruit/Adafruit_CircuitPython_BLE #2: UART ECHO TEST failure

jerryneedell, 2019-01-11:
@dhalbert do you want these type issued here or aginase CP?
I think I can repodce this  -- at least it did it twice when I tried to send more than 24 characters
See attached screenshot - it failed during the third line

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5-120-g573c6d60b on 2019-01-11; Particle Argon with nRF52840
>>> 
>>> 
>>> import ble_uart_echo_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_uart_echo_test.py", line 10, in <module>
  File "adafruit_ble/uart.py", line 72, in write
  File "adafruit_ble/characteristic.py", line 110, in value
OSError: Failed to notify or indicate attribute value, err 1304x
>>> 
```

dhalbert, 2019-01-23:
I added a FIFO to the UART, added some locking, and now it's `UARTServer`, so worth trying again.

jerryneedell, 2019-01-23:
I am gettin gthis failure when I send anything to it ```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5-207-gf46ec0253 on 2019-01-22; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> 
>>> import ble_uart_echo_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_uart_echo_test.py", line 14, in <module>
TypeError: can't convert bytes to int
>>> 
```

jerryneedell, 2019-01-23:
changing the last line to just send the byte works 
```
from adafruit_ble.uart import UARTServer

uart = UARTServer()
uart.start_advertising()

# Wait for a connection
while not uart.connected:
    pass

while uart.connected:
    # Returns b'' if nothing was read.
    one_byte = uart.read(1)
    if one_byte:
        uart.write(one_byte)
```

but it  now fails as before for long messages -- now I get up to 27 bytes before it fails

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5-207-gf46ec0253 on 2019-01-22; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> import ble_uart_echo_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_uart_echo_test.py", line 14, in <module>
  File "adafruit_ble/uart.py", line 130, in write
OSError: Failed to notify or indicate attribute value, err 1304x
>>> 
```

jerryneedell, 2019-01-23:
FYI - a few other notes:
It exits the program when I disconnect -- is that expected?

I am unable to connect to it via my iPhone or Android device -- 
The Android phone connects, but can't initialize its uart connection

so far only can connect from my desktop client  (Adafruit BlueFruit Connect)

jerryneedell, 2019-01-23:
hmm  -- any ice what the uart.write fails for me -- on a feather_nrf52840_express -- what board are you using?

jerryneedell, 2019-01-23:
which iPhone app are you using?

jerryneedell, 2019-01-23:
the iPhone does recognize that it has UART service.

jerryneedell, 2019-01-23:
I see that the usage example does not use "bytes"
```from adafruit_ble.uart import UARTServer

uart_server = UARTServer()
uart_server.start_advertising()

# Wait for a connection.
while not uart_server.connected:
    pass

uart_server.write('abc')
```

this works to my desktop client, but still not with my iPhone

jerryneedell, 2019-01-23:
just a recap of results   -- this code works fine ```from adafruit_ble.uart import UARTServer

uart = UARTServer()
uart.start_advertising()

# Wait for a connection
while not uart.connected:
    pass

while uart.connected:
    # Returns b'' if nothing was read.
    one_byte = uart.read(1)
    if one_byte:
        uart.write(one_byte)
```
the use of "bytes" in the last line examples/ble_uart_echo_test.py does not work.

For me this only work with the Adafruit BueFruit LE Connect  app on my iPhone 6 (it has IOS 12.1 installed)  My iPhone 7 with (IOS 12.1.3) does not work (also IOS 12.1.2 did not work for me on the iPhone 7.

I cannot get any Andriod version of the Adafruit BLE Connect to work yet.

jerryneedell, 2019-01-24:
@dhalbert Pulled this PR and testes with iPhone 7 IOS 12.1.3  --works fine for UART Echo and for cricket Controller demo!!! Nice work!

It still does not work with my Andriod phone -- The app connects, but the UART connection fails with the message "Uart protocol can not be initialized"

closed 2019-01-24 by tannewt