adafruit/Adafruit_CircuitPython_BLE #7: uuid module missing

jerryneedell, 2019-01-23:
I am unable to test the ibeacon since the uuid.py module is missing 
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5-207-gf46ec0253 on 2019-01-22; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> import ibeacon
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ibeacon.py", line 2, in <module>
ImportError: no module named 'adafruit_ble.uuid'
>>> import ibeacon
```

jerryneedell, 2019-01-23:
FYI - teh ble_eddystone_test works well

dhalbert, 2019-01-23:
The example in the doc is outdated. Use `from bleio import UUID` instead. Sorry about that. I updated the `examples/` but forgot about the examples in the code documentation. Will fix.

jerryneedell, 2019-01-23:
Thank -- that works!s

dhalbert, 2019-01-23:
reopening to remind me to fix the doc