adafruit/Adafruit_CircuitPython_BLE #10: UARTServer does not handle writes of >20 bytes

dhalbert, 2019-03-30:
`UARTServer.write` just writes a the TX Characteristic value directly. If the size of the byte buffer is too large (> 20 bytes), the write will fail. It should do the write in a buffered way to allow longer  buffers to be written.

Thanks to @dastels for uncovering this.

dastels, 2019-04-22:
I get this after 24 characters: 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_test.py", line 18, in go
  File "adafruit_logging.py", line 179, in info
  File "adafruit_logging.py", line 161, in log
  File "ble_handler.py", line 68, in emit
  File "/lib/adafruit_ble/uart.py", line 128, in write
OSError: Failed to notify or indicate attribute value, err 0x0013
```

dhalbert, 2019-04-22:
Could you point to the test code? Also, what are you using to receive on the client side? If it's Bluefruit LE Connect, which hardware are you running it on?

dastels, 2019-04-22:
bluefruit app on iOS

Files aren't checked in anywhere at the moment:

```import time
import random
from ble_handler import BLEHandler
import adafruit_logging as logging

l = logging.getLogger('ble')

l.addHandler(BLEHandler())

def go():
    while True:
        t = random.randint(1, 5)
        if t == 1:
            print('debug')
            l.debug("debug message %d", random.randint(0, 1000))
        elif t == 2:
            print('info')
            l.info("info message %d", random.randint(0, 1000))
        elif t == 3:
            print('warning')
            l.warning("warning message %d", random.randint(0, 1000))
        elif t == 4:
            print('error')
            l.error("error message %d", random.randint(0, 1000))
        elif t == 5:
            print('critical')
            l.critical("criticalmessage  %d", random.randint(0, 1000))
        time.sleep(5.0 + (random.random() * 5.0))```

and the handler:

```"""
UART based message handler for CircuitPython logging.

Adafruit invests time and resources providing this open source code.
Please support Adafruit and open source hardware by purchasing
products from Adafruit!

Written by Dave Astels for Adafruit Industries
Copyright (c) 2018 Adafruit Industries
Licensed under the MIT license.

All text above must be included in any redistribution.
"""

#pylint:disable=missing-super-argument

# Example:
#
# import board
# import busio
# from uart_handler import UartHandler
# import adafruit_logging as logging
#
# uart = busio.UART(board.TX, board.RX, baudrate=115200)
# logger = logging.getLogger('uart')
# logger.addHandler(UartHandler(uart))
# logger.level = logging.INFO
# logger.info('testing')

from adafruit_logging import LoggingHandler
from adafruit_ble.uart import UARTServer

class BLEHandler(LoggingHandler):
    """Send logging output to the BLE uart port."""

    def __init__(self):
        """Create an instance.

        :param uart: the busio.UART instance to which to write messages

        """
        self._advertising_now = False
        self._uart = UARTServer()
        self._uart.start_advertising()

    def format(self, level, msg):
        """Generate a string to log.

        :param level: The level at which to log
        :param msg: The core message

        """
        return super().format(level, msg) + '\r\n'

    def emit(self, level, msg):
        """Generate the message and write it to the UART.

        :param level: The level at which to log
        :param msg: The core message

        """
        while not self._uart.connected:
            pass

        data = bytes(self.format(level, msg), 'utf-8')
        for i in range(len(data)):
            self._uart.write(data[i:i+1])```

dastels, 2019-04-22:
Let me try sending the entire string at once.

dastels, 2019-04-22:
Writing the entire string immediately results in this with no characters sent.

```Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_test.py", line 24, in go
  File "adafruit_logging.py", line 197, in error
  File "adafruit_logging.py", line 161, in log
  File "ble_handler.py", line 67, in emit
  File "/lib/adafruit_ble/uart.py", line 128, in write
OSError: Failed to notify or indicate attribute value, err 0x000c
```

dhalbert, 2019-04-22:
I am doing this:
```
>>> s
bytearray(b'abcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqr')
>>> uart.write(s)
```
and it's working. Can you try that? I'll try your code too. Make sure if there are multiple copies of the library that the right one is being imported. Thanks.

dastels, 2019-04-22:
To confirm, all I need to update is adafruit_ble.uart ??

dhalbert, 2019-04-22:
That should be true, but you may as well copy the whole dir. I removed it from lib and copied my repo version to `/`. Note that you need to pull from `dhalbert/Adafruit_CircuitPython_BLE`, not `adafruit/...`, since it isn't merged yet.

dastels, 2019-04-22:
OK.. let me try again.

dastels, 2019-04-22:
Same result.

dhalbert, 2019-04-22:
@dastels: change the last lines in ble_handler to:
```
        data = bytes(self.format(level, msg), 'utf-8')
        self._uart.write(data)
```

There is some problem writing many small strings in a row, which I still need to debug, but this will work for now.

dhalbert, 2019-04-23:
OK, after more debugging, the code in the previous comment should be fine, if you send the string all at once (it will be sent in 20-byte chunks). If you send the data one byte at a time, the central will fall behind in processing and/or the maximum possible data rate will be exceeded. So this is not an bug, but indicates some throttling is needed.

If you see errors on very long messages, then some delay should be introduced.

dastels, 2019-04-23:
I’m out of the office today but will dig in tomorrow. What version of CP are you using?

dhalbert, 2019-04-23:
Absolute latest. I have some ideas about doing more flow control in CircuitPython. I am going to go ahead and merge #11 because it's clearly a fix for some of the problems. This issue will close but go ahead and continue to comment or open a new issue in the circuitpython repo if needed.

closed 2019-04-23 by dhalbert

dastels, 2019-04-24:
OK. Latest lib bundle and RC1. It "works".  No crashes, and all characters get sent.  However it comes out in chunks, assumedly because it's sending 20 chars at a time.

dhalbert, 2019-04-24:
Yes, it is sending the maximum possible in a single Bluetooth packet. There's a lot more overhead if we send a character at a time. The next step is to do some flow control on the CircuitPython side.

PR #11 (2019-04-23): Write in chunks of 20 bytes or less
Fixes #10.

@dastels Could you test this? The sending now appears to be correct: it will send no more than 20 bytes at a time.

A BLE notification is sent after each write. If the receiver does not handle the writes fast enough, bytes will be dropped. I tested by sending 200 bytes to the Bluefruit LE Connect app, and it received all 200. But a slower receiver may need to implement some kind of flow control.