adafruit/Adafruit_CircuitPython_BLE #12: Circuitpython in HASS

Tasm-Devil, 2019-05-03:
Are there any plans to support this bluetooth to mqtt gateway [bt-mqtt-gateway](https://github.com/zewelor/bt-mqtt-gateway) or any other software to use a circuitpython device in a home automated system.

I've no idea how this Bluetooth stuff works. Characteristics, Service, Descriptor puhh. 
Otherwise i would start writing a so called 'worker' for the _bt-mqtt-gateway_. 

It would be awesome to embed a circuitpython device into an existing [HASS](https://www.home-assistant.io/hassio/installation/) System.

ladyada, 2019-05-03:
hiya please get any details for the BTLE characteristics, or ask them to provide the interface they expect

Tasm-Devil, 2019-05-04:
OK. But are there any attempts from adafruit to provide a kind of gateway between circuitpython <-> btle <-> getaway (rpi) <-> mqtt <-> adafruitio (e.g.) ?

A kind of tunneled socket (over btle) to circuitpython would me even nicer.

ladyada, 2019-05-04:
not yet! if you get somewhere with it let us know :)

Tasm-Devil, 2019-05-04:
Meybe this is the way we should go: https://github.com/olliephillips/webbleMQ

Tasm-Devil, 2019-05-05:
I have now ssl on my own mosquitto server and webbleMQ works like a charm. You can try it here:
[tasm-devils webbleMQ](https://tasm-devil.github.io/webbleMQ/)

This is the most easy way to bridge BLE UART to your mqtt server.

Tasm-Devil, 2019-05-07:
https://github.com/adafruit/circuitpython/issues/1843

closed 2019-05-07 by Tasm-Devil