adafruit/Adafruit_CircuitPython_BLE #13: Peripheral name length limit?

Arudinne, 2019-06-02:
Hello,

I tried editing line 68 in uart.py to set the name as per this link:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/bleio/Device.html

'''
self._periph = Peripheral((nus_uart_service,), name='1234556789')
'''

However regardless of what app I try to use to connect it looks like it's truncating the name to 5 characters. Example:

[image]

Is this a way around this limit?

dhalbert, 2019-06-02:
In the original advertising packet, there's not enough room for the complete device name. The packet is only 31 bytes, and we need to include the service UUIDs, etc.

The complete name can be sent in a scan response packet: The scanner receives an advertisement, requests a scan response, and then further information is sent in the response, including the full name. But we haven't implemented scan responses yet. I'm working on this code right now and will be doing this soon. The original implementation was just enough to get minimal BLE peripheral support working to communicate, and there's a lot more to do. The API's may be changing slightly as we add more features. We'll have a beta with more functionality in the relatively near future.

Arudinne, 2019-06-02:
OK, well I guess that answers that then. Thanks!

closed 2019-06-02 by Arudinne