adafruit/Adafruit_CircuitPython_BLE #24: Hang when disconnecting quickly

kattni, 2019-09-26:
When one side of the connection disconnects abruptly, the code can hang with the following error:
```
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "adafruit_ble/uart_server.py", line 142, in write
OSError: Failed to read CCCD value, err 0x3002
```
Requires soft reset to continue.

Mentioned to @dhalbert who suggested filing an issue here, though it is still unclear whether or not this should be handled within this library or within CircuitPython itself.

kattni, 2019-10-02:
I walked away from the CPB with my mobile, and received a similar but slightly different error:
```
Traceback (most recent call last):
  File "code.py", line 28, in <module>
  File "adafruit_ble/uart_server.py", line 142, in write
OSError: Failed to notify or indicate attribute value, err 0x3002
```
Including here so we can cover all the bases necessary when including a fix for this.

devoh747, 2019-10-09:
You can also get a 

Traceback (most recent call last):
File "code.py", line 241, in
File "/home/travis/build/adafruit/Adafruit_CircuitPython_Bundle/libraries/helpers/bluefruitconnect/adafruit_bluefruit_connect/packet.py", line 128, in from_stream
File "/home/travis/build/adafruit/Adafruit_CircuitPython_Bundle/libraries/helpers/bluefruitconnect/adafruit_bluefruit_connect/packet.py", line 89, in from_bytes
ValueError: Bad checksum

if you send too many commands.. if you wrap the getting of a pack with a try it fixes it. I'll make a PR.

dhalbert, 2019-12-08:
Not seen with reworked code and API.

closed 2019-12-08 by dhalbert