adafruit/Adafruit_CircuitPython_BLE #27: ValueError: Bad checksum

devoh747, 2019-10-10:
Traceback (most recent call last):
File "code.py", line 241, in
File "/home/travis/build/adafruit/Adafruit_CircuitPython_Bundle/libraries/helpers/bluefruitconnect/adafruit_bluefruit_connect/packet.py", line 128, in from_stream
File "/home/travis/build/adafruit/Adafruit_CircuitPython_Bundle/libraries/helpers/bluefruitconnect/adafruit_bluefruit_connect/packet.py", line 89, in from_bytes
ValueError: Bad checksum

dhalbert, 2019-12-01:
This is still present with the new 3.0.0 API. @katnni was able to generate it.

dhalbert, 2019-12-08:
This is due to overflowing the peer's buffer, which is is a ring buffer. There is no flow control, so either the sender should regulate the flow, or the receiver should ignore bad packets.

closed 2019-12-08 by dhalbert