adafruit/Adafruit_CircuitPython_BLE #28: bleio

tcpipchip, 2019-10-15:
[image]
and i am doing something wrong ?

tcpipchip, 2019-10-15:
Thank you so much!!!!!

tannewt, 2019-10-15:
I'm changing all of this so I wouldn't spend too much time on it.

dhalbert, 2019-10-15:
I am not seeing this issue. I am scanning several beacons and am seeing the manufacturer data. Could you do something like this:
```
>>> for e in entries:
...     print(e.advertisement_packet.packet_bytes, e.name, e.address, e.manufacturer_specific_data)
...     
...     
... 
bytearray(b'\x02\x01\x1a\x02\n\x0c\x0b\xffL\x00\x10\x06\x13\x1e\x07\xd3\xa0\xcc') None <Address 71:7c:72:c4:05:81> bytearray(b'L\x00\x10\x06\x13\x1e\x07\xd3\xa0\xcc')
bytearray(b'\x02\x01\x1a\x02\n\x0c\n\xffL\x00\x10\x05\x03\x18\r\x8e\xd4') None <Address 6a:32:d7:67:aa:d8> bytearray(b'L\x00\x10\x05\x03\x18\r\x8e\xd4')
bytearray(b'\x02\x01\x1a\x02\n\x0c\n\xffL\x00\x10\x05\x03\x1c\xf4\x1a\xba') None <Address 59:1d:1f:10:67:7a> bytearray(b'L\x00\x10\x05\x03\x1c\xf4\x1a\xba')
bytearray(b'\x02\x01\x06\x1a\xff\xff\xff\x02\x15\x124Vx\x124\x124\x124\x124Vx\x9a\xbc\x00{\x00\xea\xca') None <Address f0:1f:1a:88:3e:28> bytearray(b'\xff\xff\x02\x15\x124Vx\x124\x124\x124\x124Vx\x9a\xbc\x00{\x00\xea\xca')
bytearray(b'\x02\x01\x06\x03\x03\xed\xfe') None <Address c6:d9:11:69:d9:87> None
```

(Note that you can paste code and then use the `<>` button to surround it in triple backquotes, instead of using screenshots.)

tcpipchip, 2019-10-15:
i wil test now with your code...wait...

tcpipchip, 2019-10-15:
and works fine on NRF52 SCANNER (APP)

tcpipchip, 2019-10-16:
hi, any progress ?

dhalbert, 2019-10-25:
We're redoing the API and Python library. Currently we don't handle scan responses, which might be the reason you are not seeing all the advertising data. That is changing in the upcoming release.

tcpipchip, 2019-10-25:
Thank you so much! 
I will wait!

dhalbert, 2019-12-08:
@tcpipchip If you'd like, please re-test with CircuitPython 5.0.0-beta.0 and adafruit_ble 3.1.0 or later (3.1.0 will be released soon). Closing for now, but please re-open or open a new issue if you see a problem.

closed 2019-12-08 by dhalbert

tcpipchip, 2019-12-12:
@dhalbert 

Hi, i am having problems on NINA B302 with new CIRCUITPYTHON version

I am using bootloader
feather_nrf52840_express_bootloader-0.2.13_s140_6.1.1

=====================
WORKS, I CAN ACCESS CIRCUITPY (E:) if use
adafruit-circuitpython-sparkfun_nrf52840_mini-pt_BR-4.1.0

WORKS, I CAN ACCESS CIRCUITPY (E:) if use
adafruit-circuitpython-sparkfun_nrf52840_mini-en_US-5.0.0-alpha.4

FREEZEE CIRCUITPY (E:) WHEN TRY TO ACCESS THE E: if use
adafruit-circuitpython-sparkfun_nrf52840_mini-en_US-5.0.0-beta.0

FREEZEE CIRCUITPY (E:) WHEN TRY TO ACCESS THE E: if use
adafruit-circuitpython-sparkfun_nrf52840_mini-en_US-5.0.0-beta.0

Then i cant continue the tests with BLE :(

dhalbert, 2019-12-12:
@tcpipchip This is a new bug, and it's in CircuitPython itself, not the library, so normally, please open a new issue, in https://github.com/adafruit/circuitpython. However, this is probably the same bug ashttps://github.com/adafruit/circuitpython/issues/2339 and is fixed after beta.1 by https://github.com/adafruit/circuitpython/pull/2376.

tcpipchip, 2019-12-12:
Can i use the version 4.0 to test the new ble version ?

dhalbert, 2019-12-12:
No, sorry, the API has changed a lot. You can wait for the next beta, or here is a test .uf2 to try for now, but there are still more bugs to fix: 
[sparkfun_nrf52840_mini-volatile-fix.zip](https://github.com/adafruit/Adafruit_CircuitPython_BLE/files/3957097/sparkfun_nrf52840_mini-volatile-fix.zip)

tcpipchip, 2019-12-12:
@dhalbert 
This firmware make the E: works :)
Where is the BLE LIB to me test ?

dhalbert, 2019-12-12:
https://github.com/adafruit/Adafruit_CircuitPython_BLE/releases/latest

tcpipchip, 2019-12-12:
@dhalbert 
I had found!
Good news
[image]