adafruit/Adafruit_CircuitPython_BLE #44: apple notifications - soft device error

jerryneedell, 2019-12-11:
I tried the ble_apple_notifications.py demo on a CPB with 5.0Beta.1 
this error occurred

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Circuit Playground Bluefruit with nRF52840
>>> import ble_apple_notifications
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_apple_notifications.py", line 15, in <module>
  File "adafruit_ble/__init__.py", line 174, in start_advertising
_bleio.BluetoothError: Unknown soft device error: 0007
>>> 

````

dhalbert, 2019-12-11:
Which model of iPhone or iPad are you using?

jerryneedell, 2019-12-11:
This was before I tried to connect.
I gave an iPhone 7 and an iPad pro

jerryneedell, 2019-12-11:
Does this work for you?

tannewt, 2019-12-11:
0007 is invalid parameter.

@jerryneedell can you try running it as code.py? There may be something that is being long lived. However, long living usually happens after an import finishes I believe.

jerryneedell, 2019-12-11:
@tannewt same failure as code.py on feather_nrf52840_express

jerryneedell, 2019-12-11:
I'm just puzzled what is different from my setup that yours.

jerryneedell, 2019-12-11:
just to be sure, I tried entering the commands at the REPL -- same result
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> 
>>> import time
>>> import adafruit_ble
>>> from adafruit_ble.advertising.standard import SolicitServicesAdvertisement
>>> from adafruit_ble.services.apple import AppleNotificationService
>>> 
>>> radio = adafruit_ble.BLERadio()
>>> a = SolicitServicesAdvertisement()
>>> a.complete_name = "NotifyPlease"
>>> a.solicited_services.append(AppleNotificationService)
>>> radio.start_advertising(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ble/__init__.py", line 174, in start_advertising
_bleio.BluetoothError: Unknown soft device error: 0007
```

tannewt, 2019-12-12:
I think this might be because of the new auto-scan response which adds a different complete name. Testing it now.

tannewt, 2019-12-12:
It is more complicated than that. Still looking.

commit 434e7e02 (2019-12-12): Merge pull request #45 from tannewt/fix_ancs_example

Fix ANCS example. Fixes #44
commit 53f6352e (2019-12-12): Fix ANCS example. Fixes #44
closed 2019-12-12 by dhalbert