adafruit/Adafruit_CircuitPython_BLE #46: ble_hid_periph.py does not work for me

kohlerm, 2019-12-18:
https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_hid_periph.py
does not work for me because the Keyboard constructor does not expect an argument.
I checked the library and it does not seem to be support that argument. 
Anything I'm missing here?

kohlerm, 2019-12-18:
Looks like the change to support the argument was reverted. Anyone knows why?

tannewt, 2019-12-18:
Oops! I tested with a version I had changed. Here is a WIP branch: https://github.com/tannewt/Adafruit_CircuitPython_HID/tree/pass_in_devices

I'll test it after lunch and then make a PR if it all looks good. Thanks for filing an issue!

tannewt, 2019-12-18:
PR pending here: https://github.com/adafruit/Adafruit_CircuitPython_HID/pull/43

tannewt, 2019-12-18:
Confirmed ble_hid_periph.py works with it as well.

kohlerm, 2019-12-19:
Great let me check that on the weekend :-)

tannewt, 2020-02-07:
This is fixed by https://github.com/adafruit/Adafruit_CircuitPython_HID/pull/43

closed 2020-02-07 by tannewt