adafruit/Adafruit_CircuitPython_BLE #52: Unknown soft device error: 0012 (when trying to connect multiple devices)

tadakado, 2020-01-12:
This is really cool project! I want to connect multiple device.
But, I can't connect 3 or more devices. Here is a code and error message.
I tried CurrentTimeService and ANCS, and both give the same error message.
```
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import SolicitServicesAdvertisement
from adafruit_ble.services.standard import CurrentTimeService

r = BLERadio()

r.start_advertising(SolicitServicesAdvertisement(CurrentTimeService))

# after tapping this device name on the 1st iOS
r.connections[-1].pair()

r.start_advertising(SolicitServicesAdvertisement(CurrentTimeService))

# after tapping this device name on the 2nd iOS
r.connections[-1].pair()

#### this advertising fails ###
r.start_advertising(SolicitServicesAdvertisement(CurrentTimeService))

# after tapping this device on the 3rd iOS
r.connections[-1].pair()
```
```
>>> r.start_advertising(SolicitServicesAdvertisement(CurrentTimeService))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ble/__init__.py", line 174, in start_advertising
_bleio.BluetoothError: Unknown soft device error: 0012
```

I use REPL on
feather_nrf52840_express_bootloader-0.2.13_s140_6.1.1.zip
adafruit-circuitpython-bundle-5.x-mpy-20200109.zip

dhalbert, 2020-01-12:
The number of BLE connections is fixed at compile time. Right now it's set at 2. This is because RAM allocation for the storage needed for the underlying "SoftDevice" library from Nordic that we use is also fixed, and is related to the number of connections.

We should probably give a better error. Do you have a use case for more connections, or are you trying things out? If you want to have lots of device communicating simulateneously, using advertisements rather than connections might make sense, given the limited resources available on the chip.

tadakado, 2020-01-12:
I'd like to make a centralized notification system, which captures ANCS of my family's iPhones. At this point, I have 3 iPhones. That's why I tried more than two. Also, if it can transfer aggregated notifications to a server by BLE would be nice. Temporally, I can use two or more boards and transfer by wire. Thanks!

closed 2020-01-12 by tadakado

alexwhittemore, 2020-09-16:
@dhalbert Does that two-connection limit apply regardless of platform, or is this different on the Pi with adafruit-blinka-bleio? I've got a fleet of feathers all trying to connect to a raspberry pi. The feathers all advertise a UART service, and the Pi is supposed to connect to any suitable devices it sees and sink their data.

I modified the "echo" example on the Pi side to "look for UART servers and connect," but not write data. But the whole concept seems to go awry with multiple devices. I'm not quite sure how best to juggle the simultaneous scanning and connection handling in a while-loop architecture, but I suppose maybe I should scrap the plan entirely if I'll only ever achieve two feathers anyway. Or maybe advertising just makes a lot more sense and I'm wrong that UART is the low-hanging fruit.

For reference, this is my current Pi code:

``` Python
import time
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

ble = BLERadio()
while True:
    if ble.connected and any( # This used to be "while ble.connected..." but if we do that, we'll never scan again...
        UARTService in connection for connection in ble.connections
    ):
        for connection in ble.connections:
            print(connection)
            if UARTService not in connection:
                continue
            uart = connection[UARTService]
            one_byte = uart.read()
            if one_byte:
                print(one_byte)
            print()
        #time.sleep(1)

    print("disconnected, scanning")
    for advertisement in ble.start_scan(ProvideServicesAdvertisement, timeout=1):
        if UARTService in advertisement.services and "MyFeather" in advertisement.complete_name:
            print(advertisement.complete_name)
            ble.connect(advertisement)
            #break
    ble.stop_scan()
```

tannewt, 2020-09-16:
@alexwhittemore Please open a new issue on https://github.com/adafruit/Adafruit_Blinka_bleio

The limit discussed here is for CircuitPython, not the Pi.