adafruit/Adafruit_CircuitPython_BLE #62: BLE HID paired connections not working on some Windows 10 peripherals

dhalbert, 2020-01-17:
Copied from adafruit/circuitpython#1050.

@urish: 
>Windows 10 does see the device (even though not always - I had to remove the scan_response part from the advertising so that it show it on the list of pairable devices), but disconnects about one second after pairing with a "Driver error" message:

>[image]

>Android works well

@mscosti:
>@urish I'm seeing the same thing as you. At first I thought this was related to changes I have made in an attempt to implement a different HID device incorrectly / missing something (Microsoft Radial Controller / Surface Dial) , But after seeing your report I loaded up this BLE HID Keyboard example verbatim and am seeing the same behavior on Windows 10. Same thing, where it takes a second or two for it to settle with "Driver error".

>[image]

>Android Recognizes the keyboard and am able to type on it as expected

mscosti, 2020-01-17:
ok, so, weird. Used 3.1.2 from the release page of the BLE lib instead of from the bundle and i am getting flags (but sometimes not TX power? maybe a fluke).

I was still getting the same driver issue unfortunately, but on the plus, the devices was showing up in the scans from my laptop much more reliably.

So, maybe the win10 driver issue isn't only advertisement related? I looked into the services data from nRF mobile app that i'm getting from arduino vs CPy and the difference at a glance is that with Cpy i'm seeing an 'unknown' service with a really long UUID that begins with adaf (coincidence, or is it actually adafruit shortened for some reason?). Screenshot below of top level view, I can get more screenshots of the details of the service if that is also useful

Python services after connecting to phone from nRF mobile app
[image]

Arduino services after connecting to phone from nRF mobile app
[image]

dhalbert, 2020-01-17:
This was a confluence of two separate bugs: see #63. The flags were not included in advertisements, _except_ that if you printed the advertisement (which the code I was using did, for debugging purposes), another bug cause the flags to be included.

mscosti, 2020-01-17:
That explains why running your code on 3.1.2 had advertisements for me.

Thoughts on above finding about the unknown service?

mscosti, 2020-01-17:
Just loaded it up and used my example file that doesn't have the same logging as you.

✅ am now seeing flags and tx power being sent in Advertisement
✅ no issue getting windows to see the device (before would take several reboots of board + luck)
⛔️ My machine running windows 10 still shows driver error after connecting.

Hmmm. Do you think its possible Windows is really picky and not liking the fact that an unknown service (to it) is being sent? Since its an HID profile, it may be expecting to only have known services sent? I'm just making guesses, since its the only visible difference to me between the BLE data being sent from Arduino vs Cpy.

Since you said 'CircuitPython _boards_' i'm assuming the code for that lives in the firmware itself? If we think its a valid assumption that the adafruit service is causing the issue, would it be possible to get a test build that doesn't send that out?

dhalbert, 2020-01-17:
Hmm. I'm not sure why it's still complaining. Did you remove the device in Windows before re-pairing? I don't think it should care about the extra service: I wouldn't expect it to give such a low-level error; at worst I'd expect it to refuse to do HID. I'll check among my laptops to see if I have one with builtin BLE.

If you use my example, do you _not_ get an error? Could you post your own code?

dhalbert, 2020-01-17:
Which board are you using? I'll give you a build with it turned off.

dhalbert, 2020-01-17:
Also, could you give more details of the exact builtin Bluetooth device by looking it up in Device Manager?

mscosti, 2020-01-17:
* Yeah, I have been making sure to remove it before attempting to repair again. and yeah, it does not get an error with the arduino example
* I'm using the CircuitPlayground Bluefruit
* I think the name of my Bluetooth device is 'Marvel AVASTAR'. Screenshot dump below, since i'm not sure what is most helpful

[image]

[image]

[image]

mscosti, 2020-01-17:
Not sure if relevant, but computer OS specs:
* Windows 10 Pro
* Version 1909
* OS version: 18363.535

dhalbert, 2020-01-17:
Here are two CPB builds. One is the latest tip of master, the other is the same, but with the ADAF services turned off. Not tested with HID.
[cpb-8b6133393.uf2.zip](https://github.com/adafruit/Adafruit_CircuitPython_BLE/files/4074838/cpb-8b6133393.uf2.zip)
[cpb-8b6133393-dirty-no-adaf-service.uf2.zip](https://github.com/adafruit/Adafruit_CircuitPython_BLE/files/4074839/cpb-8b6133393-dirty-no-adaf-service.uf2.zip)

dhalbert, 2020-01-17:
This is interesting: https://winbuzzer.com/2019/11/19/microsoft-sends-out-driver-update-for-older-surface-products-xcxwbn/
In particular, there's an update for the Marvell AVASTAR Bluetooth Radio Adapter.

Also zephyr was having trouble re-pairing with the Marvell: https://github.com/zephyrproject-rtos/zephyr/issues/14044.

mscosti, 2020-01-17:
First, I tried `cpb-8b6133393-dirty-no-adaf-service.uf2.zip` . I think there might be an issue with the build. After attempting to pair from pc or android, it was hanging and failing, and CPB code crashed. Weird thing is the USB device seemed to also crash and it d/c itself.

Then, I tried the master build `cpb-8b6133393.uf2.zip`, and, that seems to be working! Must mean that something between beta 3 and master fixed it 🤷‍♂ ? Glad its seems to be fixed in latest, but now I'm just more curious as to what's going on here 😅 

When I get some more time i'll try out some more BLE examples, like trying out more HID profiles/reports

dhalbert, 2020-01-17:
> Then, I tried the master build `cpb-8b6133393.uf2.zip`, and, that seems to be working! Must mean that something between beta 3 and master fixed it  ?

I implemented bonding between beta.3 and master, and changed some internal state-keeping very slightly. The zephyr issue I mentioned above seems to have something to do with connection negotiation, so it may have to do with that rather the the advertisement specifically. But the advertisements definitely were buggy.

PR #63 (2020-01-21): prevent unwanted lazy advertising object instantiation
The `flags` data in connectable advertisements was not being set when it should have been, and was being set inadvertently when doing `str(advertisement)` (e.g., when printing). This caused connectable advertisements to go out without flags unless they were printed first (sort of a Heisenbug).

- Set `flags` properly in connectable advertisements
- Do not set a default value for `AdvertisingFlags`
- `Advertisement.__str__()` is now more careful not to accidentally instantiate a lazy advertising field.
- Rename `LazyField` to `LazyObjectFie [... cut]
closed 2020-01-21 by tannewt