adafruit/Adafruit_CircuitPython_BLE #71: Fails to import on nRF52840 Express

banada, 2020-02-20:
CircuitPython cannot import the adafruit_ble library.

**Platform:** Adafruit Feather nRF52840 Express (https://www.adafruit.com/product/4062)

**CircuitPython Version:** 4.x.x and 5.x.x

**Steps to Reproduce:**
Flashed firmware and copy/pasted code according the instructions in this tutorial:
https://learn.adafruit.com/ble-hid-keyboard-buttons-with-circuitpython/ble-keyboard-buttons-libraries-and-code

**Expected Result:**
Feather will become available for BLE pairing

**Actual Result:**
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 10, in <module>
  File "adafruit_ble/__init__.py", line 33, in <module>
ImportError: This release is not compatible with CircuitPython 4.x; use library release 1.x.x
```

banada, 2020-02-20:
**Note:** Bluetooth testing platform and development platform is Windows 10

dhalbert, 2020-02-20:
Which bundle or which release are you using for `adafruit_ble`? Recent versions should give that error for 4.x.x but not for 5.0.0-<whatever>.

banada, 2020-02-20:
I am testing against:
The 5.x bundle here: https://circuitpython.org/libraries
and the 5.0.0-beta.5 firmware here: https://circuitpython.org/board/feather_nrf52840_express/

banada, 2020-02-20:
Specifically:
adafruit-circuitpython-bundle-5.x-mpy-20200219
and
adafruit-circuitpython-feather_nrf52840_express-en_US-5.0.0-beta.5

dhalbert, 2020-02-20:
I do not see this problem with 5.0.0-beta.5 and the latest `adafruit_ble` from the bundle (20200220).

Make sure there is not another `adafruit_ble` in `/`. That will override anything in `lib/`. Delete the copy in `lib/` and copy it again. If you are on Windows, do an Eject on CIRCUITPY from the taskbar after doing the copy, to make sure it is completely copied. It may take Windows 10's of seconds to complete a copy.
```
Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit Circuit Playground Bluefruit with nRF52840
>>> import adafruit_ble
>>> adafruit_ble.__version__
'5.2.0'
```

banada, 2020-02-20:
> Also make sure the 5.0.0 .uf2 was copied to the BOOT drive and not CIRCUITPY by looking at the initial REPL prompt. Sometimes that happens by accident.

This was the problem. I guess I was still running 4.x firmware. Thank you so much!

closed 2020-02-20 by banada