adafruit/Adafruit_CircuitPython_BLE #77: uuid has no attribute pack_into

mak3rm1k3, 2020-04-04:
I'm trying this out for the first time on a rpi zero w, and after installing the library I keep getting an error:

`AttributeError: 'UUID' object has no attribute 'pack_into'`

Running with Python 3 here is a sample of my code:

```
import adafruit_ble
from adafruit_ble.advertising import standard
import adafruit_ble_apple_notification_center as ancs

radio = adafruit_ble.BLERadio()
a = standard.SolicitServicesAdvertisement()
a.solicited_services.append(ancs.AppleNotificationCenterService)
```

Looking more closely the error is in line 63 of uuid/__init__.py:

```
def pack_into(self, buffer, offset=0):
        """Packs the UUID into the buffer at the given offset."""
        self.bleio_uuid.pack_into(buffer, offset=offset) #<-- this is where the error is coming from
```

any advice?

mak3rm1k3, 2020-04-06:
Thanks so much for getting back to me on this. I'm pretty green on this level of python (hardware bytes level), so I'd love help adding in pack_into to the Blinka library. What exactly do you need me to add?

```
def pack_into(self, buffer, offset=0):
        """Packs the UUID into the buffer at the given offset."""
        self.bleio_uuid.pack_into(buffer, offset=offset)
```

Something like this into the UUID class?

I think I understand your second suggestion, to just inspect the type that is returned from a bytes() when assigned to advertisement_bytes? If it doesn't return bytes, though, not sure what exception to use. Any advice?

I really appreciate your help and am very excited to get this going.

tannewt, 2020-04-07:
> Thanks so much for getting back to me on this. I'm pretty green on this level of python (hardware bytes level), so I'd love help adding in pack_into to the Blinka library. What exactly do you need me to add?

Awesome! Thank you! I'm available on [our Discord](https://adafru.it/discord) too. Just @tannewt there.

> 
> ```
> def pack_into(self, buffer, offset=0):
>         """Packs the UUID into the buffer at the given offset."""
>         self.bleio_uuid.pack_into(buffer, offset=offset)
> ```
> 
> Something like this into the UUID class?

Yup! Same signature (the `def ` line) but with code inside that sets the values in the buffer. This is the C version: https://github.com/adafruit/circuitpython/blob/master/ports/nrf/common-hal/_bleio/UUID.c#L70

> I think I understand your second suggestion, to just inspect the type that is returned from a bytes() when assigned to advertisement_bytes? If it doesn't return bytes, though, not sure what exception to use. Any advice?

Not quite. You'll want to find the advertisement class and fix it so that it returns bytes. I think you'll want to change the line [here](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/adafruit_ble/advertising/__init__.py#L285) to:

```python
return bytes(encode_data(self.data_dict))
```

> 
> I really appreciate your help and am very excited to get this going.

Great! I'm happy to help you fix it and appreciate your willingness to jump in.

Kimpeh, 2020-04-21:
I also encountered this issue "AttributeError: 'UUID' object has no attribute 'pack_into'". 
Can you please advise as to what code needs to be added to the Blinka bleio library? 

Thanks!

tannewt, 2020-04-22:
@makermelissa Any chance you could add `pack_into` into Blinka_BLE?

makermelissa, 2020-05-04:
I missed the tag. I just created a new issue at https://github.com/adafruit/Adafruit_Blinka_bleio/issues/10

tannewt, 2020-05-04:
Ok, closing this in favor of https://github.com/adafruit/Adafruit_Blinka_bleio/issues/10

closed 2020-05-04 by tannewt