adafruit/Adafruit_CircuitPython_BLE #79: More thorough documentation

dhalbert, 2020-04-16:
The library could really use more thorough documentation, through more examples, overview text, a, detailed explanatory text. A Learn Guide is probably appropriate, but there should be more inline documentation too. The declarative nature of `Service` and `Advertisement` needs to be more thoroughly explained, for example.

askpatrickw, 2020-04-24:
A link to the documentation in the Readme would be nice too.

https://circuitpython.readthedocs.io/projects/ble/en/latest/

kevinjwalters, 2020-05-18:
Some other things that would be good to document and provide/cite examples:

 - I don't yet understand exactly how `prefix` works. I think this is what determines if an incoming packet matches the type of `Advertisement` - not clear if this is dependent on fields being identically ordered
 - I've not worked out yet if field order is predictable for `Advertisement` with the CP libraries. Even if the spec. regards packets as identical with different field orders it would still be useful to be able to explicitly order these, i.e. there's bound to be some semi-broken device that needs things in a specific order.

kevinjwalters, 2020-05-27:
I had this in a sub-class of `Advertisement` and this broke the parsing on received ads from `start_scan()`:

```
    def __init__(self, key_data=None, round=0):
        super().__init__()
        if key_data is not None:
            self.key_data = key_data
        if round is not None:
            self.round = round
```

Sending 

```
<RpsEncDataAdvertisement manufacturer_data=<ManufacturerData company_id=0822 data=0a 32 ff 41 42 43 44 45 46 47 48 03 31 ff 04 > >
```

got turned into

```
<RpsEncDataAdvertisement manufacturer_data=<ManufacturerData company_id=0822 data=03 31 ff 00 > >
```

on arrival. Note the final `00` bytes, that must be linked with `round=0`.

I changed it to this as a guess:

```
    def __init__(self, *, key_data=None, round=None):
```

and that fixed it. This is not intuitive and needs documenting.

dhalbert, 2020-05-28:
> * I don't yet understand exactly how `prefix` class variable works. I think this is what determines if an incoming packet matches the type of `Advertisement` - not clear if this is dependent on fields being identically ordered.

I am working on an improvement on this to fix #82. You'll give a tuple of prefixes and it will combine them into a single concatenated prefix string to pass to `_bleio`. Also there will be more doc :smiley: .

The order does not matter for matching purposes.

> * I've not worked out yet if field order is predictable for `Advertisement` with the CP libraries. Even if the spec. regards packets as identical with different field orders it would still be useful [...]

It is not predictable. So far we haven't found a device that cares.

>   * there appears to be no `__eq__` so these will return `False` even two `Advertisement`s hold the same data

Could you open an issue for to add `__eq__`? Thanks.

dhalbert, 2020-05-28:
> I had this in a sub-class of `Advertisement` and this broke the parsing on received ads from `start_scan()`:
> 
> ```
>     def __init__(self, enc_data=None, round=0):
>         super().__init__()
>         if enc_data is not None:
>             self.enc_data = enc_data
>         if round is not None:
>             self.round = round
> ```
> 
> Sending
> 
> ```
> <RpsEncDataAdvertisement manufacturer_data=<ManufacturerData company_id=0822 data=0a 32 ff 41 42 43 44 45 46 47 48 03 31 ff 04 > >
> ```
> 
> got turned into
> 
> ```
> <RpsEncDataAdvertisement manufacturer_data=<ManufacturerData company_id=0822 data=03 31 ff 00 > >
> ```
> 
> on arrival. Note the final `00` bytes, that must be linked with `round=0`.
> 
> I changed it to this as a guess:
> 
> ```
>     def __init__(self, *, enc_data=None, round=None):
> ```
> 
> and that fixed it. This is not intuitive and needs documenting.

This sounds like a bug: something is just checking falsey-ness instead of `is None`. Could you open an issue for this too, with a bit more detail or a pointer to the code? Thanks.

kevinjwalters, 2020-05-28:
Some other things that would be good to document and provide/cite examples PART DEUX:
 - description of how `start_scan()` behaves if lengthy processing/delays are added in the loop that retrieves data from it and recommendations about not doing this if appropriate.

kevinjwalters, 2020-09-04:
I mentioned how prefix matching works in https://learn.adafruit.com/rock-paper-scissors-circuitpython/advanced-game#advertisement-matching-3070937-39 - I've assumed it still checks the length field at the start of the packet but haven't checked that yet..