adafruit/Adafruit_CircuitPython_BLE #83: Expose ADAFRUIT_COMPANY_ID and MANUFACTURING_DATA_ADT

kevinjwalters, 2020-05-14:
`_MANUFACTURING_DATA_ADT` and `_ADAFRUIT_COMPANY_ID` are defined in https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/5d584576ef79ca36506e6c7470e7ac5204cf0a8d/adafruit_ble/advertising/adafruit.py#L43-L44 as `0xFF` and `0x0822`, respectively. The underscore prefix makes them private and unusable from elsewhere.

These reference values would be useful to expose from somewhere in `adafruit_ble library` for other code which is creating packets, e.g. 
[adafruit_ble_broadcastnet](https://github.com/adafruit/Adafruit_CircuitPython_BLE_BroadcastNet/blob/c6328d5c7edf8a99ff719c3b1798cb4111bab397/adafruit_ble_broadcastnet.py#L66-L67) which is currently creating its own local variables with these values.

kevinjwalters, 2020-05-27:
I'm thinking of using `sequence_number` like the one used for `AdafruitSensorMeasurement` in https://github.com/adafruit/Adafruit_CircuitPython_BLE_BroadcastNet/blob/c6328d5c7edf8a99ff719c3b1798cb4111bab397/adafruit_ble_broadcastnet.py#L84-L85

Does it make sense for its id number (`0x0003`) to be somewhere to allow developers to use it in custom `Advertisement` packets?

tekktrik, 2022-02-02:
Resolved by PR #154!

dhalbert, 2022-02-02:
In the PR, if you use one of these magic phrases, the issue will get linked and then closed when the PR is merged:
https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword

closed 2022-02-02 by dhalbert