adafruit/Adafruit_CircuitPython_BLE #84: BLE keyboard or mouse as input

kmatch98, 2020-05-15:
I’d like an example of how to receive input from an BLE keyboard or mouse as an input device. I see an example file here but it is commented out and mentions that it needs to be updated:

[Adafruit_CircuitPython_BLE/examples/ble_hid_central.py](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_hid_central.py)

kmatch98, 2020-05-17:
@dhalbert was kind enough to give some feedback on discord, I thought it would be useful to capture that here for future reference. 

————

danh: sorry, it looks like there's a lot more work to do in the original library to get this to work; this is a bigger job even for a simple demo than I thought

kmatch98: I spent some hours looking through the various libraries from the HID peripheral example thinking I could just do the reverse. But I didn’t see a clear way to get an input report from the keyboard, and no hooks in the HID keyboard code to read an input and decode. As I’m just learning about a lot of this, I wasn’t certain if it was just my own lack of knowledge or if some key bits were missing. Thanks for looking at this, and I guess I will just stay tuned. Thanks for confirming my suspicions. Should I leave the issue open?

As a temporary solution I can use the UART input with the Bluefruit app, but will be good to have the option of a standard BLE keyboard.

danh: yes, definitely. The way the BLE services are defined is not easy to understand, and we need to add some buffering on the report characteristic so we can have a FIFO for the NOTIFY data that's coming in. The hid.py code was written early on in a rewrite of the BLE library API, and it could use some revision and addition

kmatch98: Ok, I’ll stay tuned. Thanks again for your help and prompt response. As a newbie it’s helpful to get feedback before getting too frustrated with these kinds of hurdles.

dglaude, 2020-05-27:
@dhalbert Two questions and a set of argument to have this working:

(A) Do you think that this example was ever working?

If I find the date when this example was created without comment, then matching CircuitPython Firmware on that date, for a board that existed, and with library of that date... would I be able to get key from a BLE keyboard? 

(B) Do you think it is possible to make a new demo that work with current firmware and library state (or is something missing in the core/library to get there)?

(C) The argument

Beeing able to receive input from a BLE keyboard would be great.

Today with CircuitPython we can (1) Pretend to be a USB keyboard (2) Pretend to be a BLE keyboard

On the receiving side, we with a Trinked CoProcessor we can (3) have a USB host, that convert keycode to UART. (This is outside of CircuitPython but I made a FeatherWing with that).

So the missing thing is to be able to (4) have a BLE central that receive key from a BLE keyboard.

If we have that, it would become possible to create a mini CircuitPython computer, just with a CLUE and a BLE keyboard. See the cyberDÛCK project.

kmatch98, 2020-06-02:
In looking a bit more at the gaps here and working to comprehend @dhalbert comments on the notes above and the weekly call yesterday, I discovered that the [HID keyboard library](https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/master/adafruit_hid/keyboard_layout_us.py) consists of only `write` functionality but a `read` function is absent. I think this may be why dan mentioned that there are no buffers present. So I am considering that maybe the Bluetooth code may not be the issue, but it may be a general absence of HID “Host” functionality.

street-grease-coder, 2020-10-26:
there is a now very popular esp32/arduino library for this- was there any progress on the example code here?

mmartinortiz, 2022-02-06:
> there is a now very popular esp32/arduino library for this- was there any progress on the example code here?
> 
> https://github.com/T-vK/ESP32-BLE-Mouse

I arrived to this issue looking for a way to connect a keyboard to the [Adafruit Feather nRF52840 Express](https://www.adafruit.com/product/4062) via Bluetooth, so the keystrokes can be read and used by the nRF52840. The library you linked is to do the opposite: make the ESP32 behave as a Bluetooth keyboard/mouse.

mmartinortiz, 2022-02-06:
@evaherrada is there any plan to implement this ticket? Is there anything the community can do to help? I'm more skilled at Python than at Micro controllers, but maybe I can do something.

tannewt, 2022-02-07:
@mmartinortiz I'd suggest joining the Discord to help. https://adafru.it/discord I might try to do this today actually. It probably needs a feature I'm working on for the ESP32-S3

rdagger, 2022-04-05:
@tannewt Did you make any progress on the code to read a BLE keyboard?

tannewt, 2022-05-09:
@rdagger I haven't. I got distract by USB host and then we had a baby. My goal was to sort out HID with USB Host and then apply that to BLE as well.

rdagger, 2022-05-10:
@tannewt congratulations!  I was hoping to make a wireless keyboard adapter for my 1978 Bally astrocade game console.

tannewt, 2022-05-10:
> @tannewt congratulations! I was hoping to make a wireless keyboard adapter for my 1978 Bally astrocade game console.

That sounds like a neat project! I'll have to keep that in mind. I'll shoot to pick this back up with the USB host stuff.

foxy82, 2022-11-02:
Checking if any progress has been made on this in the last 6 months?

I'm really interested in using an ESP32-C3 to connect to a bluetooth remote (Consumer Control device in HID language) so that I can transmit the key presses over wifi to a separate device (picow at the moment) that will then send them to a computer - essentially transmitting the Bluetooth device over wifi to increase it's range.

tjshippy, 2023-01-19:
Here's an upvote on this topic.  I am working on using a Clue to display the position of a mouse (X,Y) and read the click events.  From all the docs I can find, the mouse and keyboard reports are very similar.  
https://forums.adafruit.com/viewtopic.php?t=197856