adafruit/Adafruit_CircuitPython_BLE #88: Gamepad not supported as BLE HID device

MatevzSe, 2020-06-07:
```
from adafruit_ble.services.standard.hid import HIDService
from adafruit_hid.gamepad import Gamepad

hid = HIDService()
gp = Gamepad(hid.devices)
```
This returns an error:
```
Traceback (most recent call last):
  File "code.py", line 5, in <module>
  File "adafruit_hid/gamepad.py", line 55, in __init__
  File "adafruit_hid/__init__.py", line 56, in find_device
ValueError: Could not find matching HID device.
```

While using keyboard/mouse works fine. Is gamepad not supported as a BLE HID device?

dhalbert, 2020-06-07:
I don't remember clearly, but this may be because we couldn't get it to work on any hosts we tried. Gamepad HID support is sporadic, and it can depend on the details of the gamepad descriptor (ranges of X and Y, number of buttons, etc.)

MatevzSe, 2020-06-07:
Can you point me in any direction, where I can find some code to maybe develop something on my own?

All in all what I am interested in is that computer or android phone recognizes device as a gamepad and sends x and y coordinates. Is this maybe achivable through Arduino code, micropython?

dhalbert, 2020-06-07:
Do you have an example of one or more existing BLE gamepads? The CircuitPython support is only Bluetooth Low Energy (BLE), not classic Bluetooht.

 I have an Xbox and controller I got for testing rasons, and though it has Bluetooth, it does not show up on nRFConnect or other BLE scanners. I believe it may be classic Bluetooth, not BLE. The same may be true of the PS4 controller, but I don't have one to try. I did some websearching and found no evidence that the PS4 controller is BLE.

dhalbert, 2020-06-08:
The transport is different: it's over Bluetooth Low Energy, or over USB. But my point is that the wireless controllers that use Bluetooth don't necessarily use Bluetooth Low Energy (BLE) , sometimes called Blueooth Smart for the radio protocol; instead they use what's called Bluetooth Classic, which is a different protocol, and is not supported by the nRF boards under CircuitPython or anything else.

MatevzSe, 2020-06-08:
So this could be achieved with classic Bluetooth, which means switching to HC-05 and not using CircuitPython, probably achievable with arduino code? Too bad Bluefruit EZ-Link is discontinued.

dhalbert, 2020-06-08:
The HC-05 provides a specific classic Bluetooth serial protocol, and cannot be used to emulate other Bluetooth devices like gamepads. Let me restate what I know and don't know:

- We didn't enable BLE Gamepad support over BLE HID. I don't remember why, but it might have been because I couldn't get it to work one or more host computers. I don't remember if I tried it on Android.
- I asked if you knew whether the Bluetooth controllers you wanted to emulate were Classic Bluetooth or BLE. My suspicion is that they are Classic Bluetooth, which we can't emulate in CircuitPython on an nRF board (nor on HC-05 or Bluefruit EZ-Link). If they were BLE, we could get their HID descriptors and reproduce those (assuming they weren't proprietary in some way).

If you know of a BLE game controller that would work for you, that is the best starting point. I don't know if BLE game controllers even exist. But there's a lot to learn here, and it might require a custom version of CircuitPython

MatevzSe, 2020-06-08:
I am not emulating any specific controller, I am trying to build my own.
Ok, let me rephrase that and explain my project. I would like to build balancing board, that would send data (computed by accel/gyro) to android/computer wirelessly, emulating movement similar to gamepad joystick x and y. The idea is that board could be used by different apps that supports controllers joystick.

closed 2020-06-30 by dherrada