adafruit/Adafruit_CircuitPython_BLE #91: CircuitPython Locks Up on Disconnect

madgrizzle, 2020-06-24:
Using the current time service example on a Adafruit Feather nRF52840 Sense with CircuitPython  5.3 (and on 5.4-beta1):

https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_current_time_service.py

I connected the sense with my android phone running the Nordic nRF Connect for Mobile app.

dhalbert, 2020-06-24:
I have tried this on a Samsung tablet, but it doesn't seem to provide the time service.

I thought that nRFConnect doesn't provide the time service, so have you paired the Sense previously with your phone via the Bluetooth settings? I'm guessing you are _not_ connecting via the Connect button in nRFConnect, but are somehow forcing a disconnect from there. Could you give a step-by-step recipe? If you could start by forgetting Sense in Android, and then going from there, that would be helpful.

madgrizzle, 2020-06-24:
I configured the GATT Server in nRFConnect to include Current Time Service (menu->Configure GATT Server->Add Service and select the Current Time Service under Select server configuration). I then connect to the sense using the connect button in nRFConnect, it then asks me if I want to pair (I did the forget as requested).  Once paired, the sense starts streaming the time.  When I close the tab for the sense, the sense locks up.  But if I press the disconnect instead of closing the tab, the time service still runs (not sure why).  If instead I swipe the nRFConnect program away (to close it), the sense crashes and provides a _bleio.BluetootError: Unknown gatt error: 0x0101 but doesn't lock up.

madgrizzle, 2020-07-03:
I can also confirm this happens using a custom built version of GadgetBridge, not just nRFConnect.  It appears that killing the connection (shutting down the app) typically results in an "unknown gatt error 0x101", which can be caught (try/exception).  But if you ruthlessly shutdown (i.e., power off the phone) or you move the phone too far away from the board, it often results in a complete lock-up of the board (requiring a hard reset).

tannewt, 2021-04-01:
@madgrizzle Please try CircuitPython 6.2.0-rc.0. I think I just fixed this. The underlying PacketBuffer wasn't tracking the disconnect correctly.