adafruit/Adafruit_CircuitPython_BLE #94: Receiving >20 bytes at once from Bluefruit Connect app not working

dhalbert, 2020-06-30:
@ Thornhill in discord says a UART message of  >20 bytes from the Bluefruit Connect app to, say, `ble_uart_echo_test`, is dropped. Sending to/from two CircuitPython boards does work. Also, does work sending to Arduino.

Confirmed by @caternuson.

caternuson, 2020-06-30:
Yep. Just adding results of my simple and brief test. Phone is Android. BLE Connect v3.3.2. Running this example:
https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_uart_echo_test.py

[image]

```python
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit ItsyBitsy nRF52840 Express with nRF52840
>>> import ble_uart
b't'
b'e'
b's'
b't'
b'\n'
b't'
b'e'
b's'
b't'
b'2'
b'\n'
```

tekktrik, 2022-06-08:
Aha!  I'm working on https://github.com/adafruit/Adafruit_CircuitPython_BluefruitConnect/issues/16 and this seems to be blocking that, given the size of images.  My testing confirms what @caternuson has described.

tekktrik, 2022-06-13:
@dhalbert and @caternuson I can also send longer UART between boards.  Should this be transferred to the app's repo?

dhalbert, 2022-06-13:
Maybe this is due to the app. I was afraid it is more general. There was this bug: https://github.com/adafruit/circuitpython/issues/3975, but it's now fixed.

dhalbert, 2024-10-17:
This is due to the default `max_length` for variable-length Characteristics being `20`. It would be fixed by #37.

I looked at a couple of implementations of NUS (Nordic UART Service). The max length of the characteristics is not fixed by the NUS definition. Instead, it's chosen at build time, based on the maximum length on the MTU size - 3.

PR #203 (2024-10-21): Make UARTService characteristics be at least as large as max MTU-3
Fixes #94.

Nordic UART Service clients expect that they will be able to send MTU-3 bytes all at once, in a single write. The current code was expecting no more than 20-byte writes.

Currently the maximum MTU-3 used by all CircuitPython ports is 512-3 (which is wrong, it should be 512). Change the `max_length` for the characteristic to be 512, and also enlarge the PacketBuffer correspondingly.

Tested with UART sends from Adafruit Bluefruit Connect app and nRF Toolbox on iOS on nRF52840 and Metro ESP32-S3. Also tested for regre [... cut]
closed 2024-10-21 by tannewt