adafruit/Adafruit_CircuitPython_BLE #106: Trying to get my feather sense to send data to my desktop, seem to be missing something

canyon289, 2020-11-05:
Hi,
I'm trying to get data from my feather sense sent to my desktop over BLE but I seem to be missing something. Currently just trying to send over the counts.

I've also recorded a video explanation to help as well.  https://youtu.be/I14jim7SfaE

I'm new to the circuit python world so I may be missing something obvious. Any tips or pointers as to what that could be would be helpful.

Thanks in advance

## Details

Here's what my `code.py` on the device looks like.
```
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService
# from adafruit_ble.uart_server import UARTServer
 
ble = BLERadio()
uart_server = UARTService()
advertisement = ProvideServicesAdvertisement(uart_server)
 
 
while True:
    ble.start_advertising(advertisement)
    print("Waiting to connect")
    while not ble.connected:
        pass
    print("Connected")
    
    i = 0
    while ble.connected:
        i += 1
        result = f"Hi from feather {i}"
        print(result)
        uart_server.write(result.encode())
        time.sleep(1)
```

I copied it from this example in this pdf https://readthedocs.org/projects/adafruit-circuitpython-bluefruitconnect/downloads/pdf/latest/

### Documentation
[image]

### Central and packet capture
On the central side, my Ubuntu desktop. I seem to be receiving the packets for pairing and disconnecting, but nothing else. I am using this command to create the packet capture
```
tcpdump -vv -i bluetooth0 -w bluetooth0_capture_gh_issue.pcap
```
[image]

Ive attached the packet dump as well in case it means anything
[bluetooth0_capture_gh_issue.pcap.zip](https://github.com/adafruit/Adafruit_CircuitPython_BLE/files/5492018/bluetooth0_capture_gh_issue.pcap.zip)

# Versions
> Adafruit BLE 7.3.1
> Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather Bluefruit Sense with nRF52840

ladyada, 2020-11-05:
hi are you following this guide?
https://learn.adafruit.com/circuitpython-ble-libraries-on-any-computer
https://learn.adafruit.com/circuitpython-ble-libraries-on-any-computer/ble-uart-example

canyon289, 2020-11-05:
Definitely, and thank you for these awesome guides!

I tried this one as it showed how to send information back from the central. Looking at the code the relevant code blocks were the same as the pdf example above which adds to my confusion! I believe you all it works, hence the confusion. But it's a new day so let me try again.

Thanks @ladyada. Appreciate the quick answer

```
from adafruit_ble.services.nordic import UARTService
uart = UARTService()
uart.write(result.encode("utf-8"))
```

[image]

canyon289, 2020-11-05:
@ladyada, and @dhalbert Thank you so much for the example and great libraries. In short it works

For those coming to this later. It seems like packet don't get captured in tcpdump. This follow the example above.

Closing as this is resolved. Much appreciated

[image]

closed 2020-11-05 by canyon289