adafruit/Adafruit_CircuitPython_BLE #113: Async waiting for reply even if Characteristic.WRITE_NO_RESPONSE specified

licryle, 2020-12-28:
TLDR: I suspect there's an OR that should be an AND in [_bleio.characteristic at line 195](https://github.com/adafruit/Adafruit_Blinka_bleio/blob/088be251df4cc5eef1f46671963dbb4124269c12/_bleio/characteristic.py#L195).

Hi folks,

I'm using adafruit_ble to connect to an off-the-shelf BLE lamp that implements a custom protocol.

To do so, I've implemented a quick custom Service using StreamIn/Out characteristics.
In them, I specify Characteristic.WRITE_NO_RESPONSE. You can see the [source code here](https://pastebin.com/Ry2eSY0a).

However, at run time, the writing hangs waiting for a reply from the lamp.
I tracked it down to [_bleio.characteristic at line 195](https://github.com/adafruit/Adafruit_Blinka_bleio/blob/088be251df4cc5eef1f46671963dbb4124269c12/_bleio/characteristic.py#L195).

However, the line states:
response=self.properties | Characteristic.WRITE

According to the documentation, [response should be a bool](https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.client.BaseBleakClient.write_gatt_char) where true means a reply is expected..
I may be mistaken, but OR seems to point as an "always true", while a AND would mean "only await a reply when config is specifically WRITE", hence I wonder if that should be a "&".

tannewt, 2020-12-28:
@dhalbert Have any context on this? My brain has purged BLE info.

dhalbert, 2020-12-28:
I will take a look!