adafruit/Adafruit_CircuitPython_BLE #119: BLE scanning timeout ignored on nRF52840

alexwhittemore, 2021-01-26:
I set up a little color sender example as a debug exercise. This is the receive side of it (The send side is commented out):

``` python
# Color Sender

import time
import board
import digitalio

import neopixel

from adafruit_ble import BLERadio
from adafruit_ble.advertising.adafruit import AdafruitColor

# The color pickers will cycle through this list with buttons A and B.
color_options = [
    0x110000,
    0x111100,
    0x001100,
    0x001111,
    0x000011,
    0x110011,
    0x111111,
    0x221111,
    0x112211,
    0x111122,
]

ble = BLERadio()

button = digitalio.DigitalInOut(board.SWITCH)
button.pull = digitalio.Pull.UP                     # Button is active LOW

led = digitalio.DigitalInOut(board.D13)
led.switch_to_output()

neopixels = neopixel.NeoPixel(board.NEOPIXEL, 10, auto_write=False)

i = 0
advertisement = AdafruitColor()
advertisement.color = color_options[i]
neopixels.fill(color_options[i])
closest_last_time = time.monotonic()
neopixels.fill(0x000000)
neopixels.show()
while True:
    # The first mode is the color selector which broadcasts it's current color to other devices.
    # print("Broadcasting color")
    # ble.start_advertising(advertisement)
    # last_i = i
    # if not button.value:            # Active low.
    #     i += 1
    # i %= len(color_options)
    # if last_i != i:
    #     color = color_options[i]
    #     neopixels.fill(color)
    #     print("New color {:06x}".format(color))
    #     neopixels.fill(0x000000)
    #     neopixels[0] = color
    #     neopixels.show()
    #     advertisement.color = color
    #     ble.stop_advertising()
    #     ble.start_advertising(advertisement)
    #     time.sleep(0.5)
    # ble.stop_advertising()
    # The second mode listens for color broadcasts and shows the color it hears.
    print("Scanning for colors")
    for entry in ble.start_scan(AdafruitColor, minimum_rssi=-100, timeout=1):
        closest_last_time = time.monotonic()
        print(entry.rssi, entry.color)
        neopixels.fill(0x000000)
        neopixels[0] = entry.color
        neopixels.show()
        ble.stop_scan()
    print("out of loop")
    # Clear the pixels if we haven't heard from anything recently.
    now = time.monotonic()
    if now - closest_last_time > 1:
        neopixels.fill(0x000000)
        neopixels.show()
    ble.stop_scan()
```

Note `ble.stop_scan()` at the end of the `for entry in ble.start_scan(AdafruitColor, minimum_rssi=-100, timeout=1):` section.

With that stop_scan() present, of course it stops scanning every advert it hears, and I get the following output:

```
Scanning for colors
-41 1114112
out of loop
Scanning for colors
-52 1114112
out of loop
Scanning for colors
```

If I remove the `ble.stop_scan()`, expecting the `start_scan()` to return after 1s, it simply never does. `print("out of loop")` just never happens.

Maybe the underlying implementation on nRF is buggy and this is actually a circuitpython issue?

alexwhittemore, 2021-01-26:
Nevermind, that last line seems to be right, I guess this is a new bug in CP 6.0.1 (which I'm using). https://github.com/adafruit/circuitpython/issues/3942

closed 2021-01-26 by alexwhittemore

dhalbert, 2021-01-26:
Yes, this is fixed in 6.1.0.

alexwhittemore, 2021-01-26:
Oh I hadn't even realized 6.1 was already released - I'll have to just go update :)