adafruit/Adafruit_CircuitPython_BLE #121: BLE HID does not work on Raspberry Pi

drewskoots, 2021-03-12:
I'm trying to port [this tutorial](https://learn.adafruit.com/ble-hid-keyboard-buttons-with-circuitpython/overview) to my raspberry pi zero, but upon running the script I get the following error:
```
Traceback (most recent call last):
  File "code.py", line 22, in <module>
    hid = HIDService()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/services/standard/hid.py", line 324, in __init__
    super().__init__(report_map=hid_descriptor)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/services/__init__.py", line 70, in __init__
    getattr(self, class_attr)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/characteristics/__init__.py", line 142, in __get__
    return bleio_characteristic.value
  File "/usr/local/lib/python3.7/dist-packages/_bleio/characteristic.py", line 183, in value
    self.service.connection._bleak_client.read_gatt_char(self.uuid._bleak_uuid)
AttributeError: 'NoneType' object has no attribute '_bleak_client'
```

This seems to be pretty far down the stack, so I'm finding it very difficult to debug. Does anyone have any tips on what could be the issue? Is it a fundamental hardware issue or can I tweak the code to make it run on my raspi0. I'm not very well versed in BLE, so I was hoping someone with more expertise could maybe offer some insight. I'd greatly appreciate any help :)

dhalbert, 2021-03-12:
Are you trying to get the Rpi to act as an HID keyboard. The RPi, using adafruit-blinka-bleio, and bleak, can only act as a BLE central, and the HID code acts as a peripheral.

drewskoots, 2021-03-12:
Yes, the idea is that I would like the raspi to send a series of keypresses to my computer when I push a button. Preferably over bluetooth rather than over USB.

dhalbert, 2021-03-12:
Unfortunately, the underlying `bleak` layer does not support that. It can't advertise, because it's a central only. See https://github.com/hbldh/bleak/issues/81. We don't know of another underlying multi-platform library we can use that would be able to act as a peripheral.

Other people have wanted to do this, but I haven't found an example of an existing project.

drewskoots, 2021-03-12:
Ok, that's too bad. I definitely appreciate you helping me get a definitive answer here though. I'll see if I can achieve the same functionality with a different approach!

closed 2021-03-12 by dhalbert