adafruit/Adafruit_CircuitPython_BLE #122: simple scan not receiving ble devices on raspberry pi 4

NR5P, 2021-03-21:
I'm running raspbian on raspberry pi 4 and simple scan example is not receiving my ble device that is advertising. I do receive it with hcitool on the raspberry pi 4 

`
pi@raspberrypi:~/Desktop/raspi_bluetooth $ sudo hcitool lescan
LE Scan ...
F4:CF:A2:8D:30:FA (unknown)
F4:CF:A2:8D:30:FA (unknown)
`

The simple scan example I'm running in python is 
`
# SPDX-FileCopyrightText: 2020 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

from adafruit_ble import BLERadio

ble = BLERadio()
print("scanning")
found = set()
scan_responses = set()
for advertisement in ble.start_scan():
    addr = advertisement.address
    if advertisement.scan_response and addr not in scan_responses:
        scan_responses.add(addr)
    elif not advertisement.scan_response and addr not in found:
        found.add(addr)
    else:
        continue
    print(addr, advertisement)
    print("\t" + repr(advertisement))
    print()

print("scan done")
`

dhalbert, 2021-03-21:
1. Are you running the Python scan example under `sudo`? Your user needs to be in the `bluetooth` group. See https://learn.adafruit.com/circuitpython-ble-libraries-on-any-computer/install-ble-libraries#linux-and-raspberry-pi-add-user-to-bluetooth-group-3074922-4
2. The linux kernel does de-duplication of advertisements, which may be dropping some or all of your advertisements. See https://github.com/adafruit/Adafruit_Blinka_bleio#support-for-duplicate-advertisement-scanning-on-linux
3. If you just print out all the advertisements you are seeing in your test program, before you prune them, are you seeing the ones you expect?

NR5P, 2021-03-27:
not running as root was the issue.

closed 2021-03-27 by NR5P