adafruit/Adafruit_CircuitPython_BLE #129: macOS Big Sur troubleshooting: Abort trap: 6

cygann, 2021-08-05:
Attempting to run a simple example on my computer (macOS Big Sur), the program would crash almost immediately with the following error message (no other information):
```
Abort trap: 6
```

After a bit of web searching, I found that I needed to add my terminal application to the Bluetooth privacy permissions under System Preferences > Security & Privacy > Privacy > Bluetooth. After doing this everything worked!

This is not an impending issue as I resolved it, but I thought it would be helpful to anyone facing a similar problem in the future.

cygann, 2021-08-05:
As noted above, this has been resolved.

closed 2021-08-05 by cygann

dhalbert, 2021-08-05:
Thanks, we will add this to the appropriate Learn Guide(s).