adafruit/Adafruit_CircuitPython_BLE #130: Reconnecting to Feather nRF52840 Express causes exception

Kimpehzor, 2021-08-14:
I am running [this example code](https://github.com/adafruit/Adafruit_CircuitPython_BLE_Apple_Notification_Center/blob/main/examples/ble_apple_notification_center_new.py) on a Feather nRF52840 Express.

```
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""
This example solicits that apple devices that provide notifications connect to it, initiates
pairing, prints existing notifications and then prints any new ones as they arrive.
"""

import time
import adafruit_ble
from adafruit_ble.advertising.standard import SolicitServicesAdvertisement
import adafruit_ble_apple_notification_center as ancs

# PyLint can't find BLERadio for some reason so special case it here.
radio = adafruit_ble.BLERadio()  # pylint: disable=no-member
a = SolicitServicesAdvertisement()
a.solicited_services.append(ancs.AppleNotificationCenterService)
radio.start_advertising(a)

while not radio.connected:
    pass

print("connected")

known_notifications = set()

while radio.connected:
    for connection in radio.connections:
        if not connection.paired:
            connection.pair()
            print("paired")

        ans = connection[ancs.AppleNotificationCenterService]
        for notification in ans.wait_for_new_notifications():
            print(notification)

        print(len(ans.active_notifications))
    time.sleep(1)

print("disconnected")
```
The script runs without issue until I encounter a reconnect. If the Bluetooth connection is disconnected (manually by disabling Bluetooth on the phone or walking out of Bluetooth range) and then reconnected, this exception is thrown upon reconnecting: 
```
Traceback (most recent call last):
File "code.py", line 33, in
File "adafruit_ble/init.py", line 88, in getitem
File "adafruit_ble/init.py", line 52, in _discover_remote
ConnectionError: Connection has been disconnected and can no longer be used. Create a new connection.

```

Restarting the script does not help. Disabling / Re-enabling bluetooth does not help.

I have reproduced the issue using:
CircuitPython - 6.3.0
CircuitPython - 7.0.0-alpha.6
iOS 14.7
iOS 15 Public Beta 5

tannewt, 2021-08-24:
Please try absolute newest of CircuitPython. I just tried the demo and it worked ok.

blalor, 2021-09-01:
I'm also experiencing this error, using CircuitPython 6.3.0 and the "Absolute Newest" bundle `adafruit-circuitpython-bundle-py-20210901.zip` (which I wouldn't expect to completely work because different major versions, but I'm trying to hack my way around this issue).

@tannewt is this issue coming from this project, or from CircuitPython itself?

blalor, 2021-09-01:
Interesting, after "forgetting" the CircuitPython device from my iPhone, the problem seems to have gone away. 🤔

tannewt, 2021-09-01:
> @tannewt is this issue coming from this project, or from CircuitPython itself?

I suspect the CircuitPython core. We reworked BLE quite a bit to add the BLE workflow in 7 so it's not really worth debugging 6.3.0. I'm planning a 7 release candidate this week and hopefully a stable release next week or the week after depending on how the RC goes.

closed 2021-09-10 by tannewt