adafruit/Adafruit_CircuitPython_BLE #131: Characteristic not binding when calling its write to stream function.

exinmusic, 2021-08-15:
I'm having trouble getting a custom service to bind upon trying to write to its tx characteristic. Looking at the trace, the issue is in bleak or bluez and I'm not sure if I'm doing something wrong in my script or if my system perhaps is missing a dependency/configuration? 

```Traceback (most recent call last):
  File "tama1.py", line 34, in <module>
    tama_connection[TAMAService].write(ln)
  File "/root/tama1/ble_services.py", line 58, in write
    self._tx.write(buf)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/characteristics/__init__.py", line 201, in __get__
    bound_object = self.bind(service)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/characteristics/stream.py", line 62, in bind
    bound_characteristic.set_cccd(notify=True)
  File "/usr/local/lib/python3.7/dist-packages/_bleio/common.py", line 607, in set_cccd
    self._notify_callback,
  File "/usr/local/lib/python3.7/dist-packages/_bleio/common.py", line 140, in await_bleak
    return future.result(timeout)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/client.py", line 927, in start_notify
    assert_reply(reply)
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.NotSupported
```
I was sure to follow each of these steps: https://learn.adafruit.com/circuitpython-ble-libraries-on-any-computer/install-ble-libraries

Hardware is an rpi zero w running 2021-05-07-raspios-buster-armhf-lite.
Script: https://gist.github.com/exinmusic/abfc39f3a7c48afd493e33b6cbe67a63
Custom Service: https://gist.github.com/exinmusic/e85e9cca226f58bf4f96b2e59515d283

Any help would be greatly appreciated. I feel like I don't have the best luck with bluetooth on rpis, probably due to my cursory bluetooth knowledge...

Again thanks for any suggestions! <3

dhalbert, 2021-08-15:
`TAMAService` looks like yet another Nordic-UART-ish service, with different UUID's. Take a look at https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/adafruit_ble/services/nordic.py and see if you can use that, and just change the UUID's. Make sure the TX and RX UUID's are correct: it's easy to get them backwards. Note in that link how they are reversed depending on whether it's central or peripheral.

exinmusic, 2021-08-15:
@dhalbert thank you for the quick response! Yes TAMAservice is made as an adaptation of nordics UART service. So taking the custom service out of the equation and running the nordic uart example against my Bluefruit LE Friend in DAT mode I yield the same error. So I dont think this is a result of the service class definition since im using the provided one from `adafruit_ble.services.nordic`.

dhalbert, 2021-08-15:
If the RX and TX are backwards, then you would get an error because the permissions on the characteristics are wrong. So you could try the `UARTService` code and just substitute the UUIDs. If it doesn't work try exchanging the RX and TX UUIDs.

exinmusic, 2021-08-16:
Yields this error if I swap the UUIDS:
```Traceback (most recent call last):
  File "tama1.py", line 39, in <module>
    replay(tama_connection, f)
  File "tama1.py", line 11, in replay
    tama_connection[TAMAService].write(ln)
  File "/root/tama1/ble_services.py", line 55, in write
    self._tx.write(buf)
AttributeError: 'CharacteristicBuffer' object has no attribute 'write'
```

Maybe a more important discovery is that I get my original error using the UARTservice echo test in the examples.

dhalbert, 2021-08-16:
There may be a BLE setup issue on the RPi. See if you are in the bluetooth group (I think there's one), and that your installation is completely up to date.

exinmusic, 2021-08-16:
Okay, so I added both root to the lp and bluetooth groups again, and to which it told me root was already in them.
So next I added pi to both groups and rebooted. Same error running echo script as either user.
ran this again just to be sure `apt upgrade bluez-firmware` and rebooted. 
Heres the bluetooth service status.

```
root@raspberrypi:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-08-16 02:59:44 BST; 1min 27s ago
     Docs: man:bluetoothd(8)
 Main PID: 360 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 725)
   CGroup: /system.slice/bluetooth.service
           └─360 /usr/lib/bluetooth/bluetoothd --compat --noplugin=sap -E

Aug 16 02:59:43 raspberrypi systemd[1]: Starting Bluetooth service...
Aug 16 02:59:44 raspberrypi bluetoothd[360]: Bluetooth daemon 5.50
Aug 16 02:59:44 raspberrypi systemd[1]: Started Bluetooth service.
Aug 16 02:59:44 raspberrypi bluetoothd[360]: Starting SDP server
Aug 16 02:59:44 raspberrypi bluetoothd[360]: Excluding (cli) sap
Aug 16 02:59:44 raspberrypi bluetoothd[360]: Bluetooth management interface 1.18 initialized
```

Whats interesting is the UART echo example if I just send it, the script will read not problem. Just errors on writing.

exinmusic, 2021-08-16:
Upgrading bluez from version 5.0 to 5.1 got me around the error.  ¯\_(ツ)_/¯
Found this article helpful, just replaced 5.0 with 5.1 in the examples: https://scribles.net/updating-bluez-on-raspberry-pi-from-5-43-to-5-50/

Closing, as this got me around the initial error I reported.
thank you @dhalbert

exinmusic, 2021-08-16:
Okay, after getting the nordic UARTservice to work, I am still facing my original error. @dhalbert I have modified the TAMAservice to be exactly the same as the nordic UARTservice in this library but with the swapped out UUIDs.
https://gist.github.com/exinmusic/e85e9cca226f58bf4f96b2e59515d283

I can write to the TAMA tx characteristic using an app, BlueSee, on my mac the same I would the UARTservice. And I've even used the old bluefruit library to perform what I'm trying to do before. Heres my old script: https://github.com/exinmusic/tama_on_research/blob/master/software/yomamagotchi.py

dhalbert, 2021-08-16:
The 128-bit UUID's you are using are the 128-bit forms of what are actually 16-bit UUIDS:
https://stackoverflow.com/questions/36212020/how-can-i-convert-a-bluetooth-16-bit-service-uuid-into-a-128-bit-uuid

In other words `0000FFF0-0000-1000-8000-00805F9B34FB` is equivalent to the 16-bit UUID `FFF0`. You could trying using the 16-bit UUIDs. Supply them as integers, not `bytes`. I'm not sure that will help, but you could try it. I think that may be the issue here.

You could also try using https://github.com/hbldh/bleak directly, which is what we're using under the covers. There are some simple examples in its documentation which would help you determine if the trouble is at its level or higher up.

I am mystified by why this should matter, but perhaps an implicit internal conversion is being done.

exinmusic, 2021-08-16:
Supplying UUID like so `StandardUUID(0xFFF0)`, `StandardUUID(0xFFF1)`,`StandardUUID(0xFFF2)`, returns the same error.
 I'll take a look at the examples and see if I have any luck with bleak directly.

exinmusic, 2021-08-16:
@dhalbert So! Using the standard 16bit UUID function instead of the vendor, AND flipping the position of the UUIDs in the TX/RX definitions got me working!

In conclusion: 
- ensuring bluez is version 5.1 instead of 5.0 (or maybe just latest? As current stable version is 5.6)
- swapping UUIDs for tx and rx
- and using StandardUUID with 16bit ids rather than VendorUUID with 128bit ids.

closed 2021-08-16 by exinmusic

dhalbert, 2021-08-16:
I am not surprised about the flipping. I am surprised about the 16 vs 128, but it may be a bleak or bluez issue. Thanks for your perserverance.