adafruit/Adafruit_CircuitPython_BLE #133: [7.0.0.beta0] Unknown system firmware error: 3205

blalor, 2021-09-01:
I'm attempting to use the AppleMediaService advertisement, per the example [here](https://github.com/adafruit/Adafruit_CircuitPython_BLE_Apple_Media/blob/1de22a217258e37434bbbe17347be5e34077291f/examples/ble_apple_media_simpletest.py).  I've run into the same problem as @Kimpehzor in #130, so per @tannewt's direction I'm trying the "Absolute Newest" CircuitPython build and bundle:

* `adafruit-circuitpython-feather_nrf52840_express-en_US-20210901-2c45f94.uf2`
* `adafruit-circuitpython-bundle-py-20210901.zip`

When `start_advertising()` is called, I get an exception:

```
Traceback (most recent call last):
  File "code.py", line 10, in <module>
  File "/lib/adafruit_ble/__init__.py", line 195, in start_advertising
_bleio.BluetoothError: Unknown system firmware error: 3205
```

I don't get this exception with the latest 6.3.0 build and bundle, but I do still get the issue in #130.

```
Adafruit CircuitPython 7.0.0-beta.0-171-g2c45f948a on 2021-09-01; Adafruit Feather nRF52840 Express with nRF52840
```

## `INFO_UF2.TXT`

```
UF2 Bootloader 0.2.6 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Dec 21 2018
```

## code

```python
import adafruit_ble
from adafruit_ble.advertising.standard import SolicitServicesAdvertisement
from adafruit_ble_apple_media import AppleMediaService

radio = adafruit_ble.BLERadio()
radio.name = "ohai"

advertisement = SolicitServicesAdvertisement()
advertisement.solicited_services.append(AppleMediaService)
radio.start_advertising(advertisement)
```

blalor, 2021-09-01:
Hm, this appears related to https://github.com/adafruit/circuitpython/issues/3731

tannewt, 2021-09-01:
Interesting! Error 0x3205 is `BLE_ERROR_GAP_DEVICE_IDENTITIES_DUPLICATE`. Does this still happen if you press reset when the LED is fast blinking blue? That'll clear your bonding information. I suspect a core problem so I'm going to move this issue to the core. It'll be easier to track then.