adafruit/Adafruit_CircuitPython_BLE #134: Unknown system firmware error: 3204 [CiPy 7.0-rc1]

rdoursenaud, 2021-09-05:
```
Adafruit CircuitPython 7.0.0-rc.1 on 2021-09-02; Adafruit Feather nRF52840 Express with nRF52840
```

```
adafruit_ble.__version__
'8.0.3'
```

```py
# --- nRF52840 BLE ---
ble = adafruit_ble.BLERadio()

print("Scanning BLE devices...")
advertisements = ble.start_scan()

# Error
print(next(advertisements))
```

```
Traceback (most recent call last):
  File "code.py", line 8, in <module>
  File "adafruit_ble/__init__.py", line 250, in start_scan
_bleio.BluetoothError: Unknown system firmware error: 3204
```

tannewt, 2021-09-07:
Thanks for trying the RC! This is due to duplicate bonding data. Please press the reset button during the blue blink at startup. That will clear existing bonding info (and make the device advertise for the CP BLE workflow.)

rdoursenaud, 2021-09-08:
Hi @tannewt,
Thanks for your insight.
So, if I understand correctly, this is a Circuitpython behavior unrelated to the BLE module.
Is there a way to do this from code? in boot.py for example?
Shouldn’t this be taken care of by the update?

rdoursenaud, 2021-09-08:
Just tested clearing the bonding info using the reset button.
Unfortunately, it only works the first time.
As soon as a device is paired, running the code again leads to the same error.
This used to work pretty well under CircuitPython 6.
I’m pretty sure I’m missing something obvious here…

tannewt, 2021-09-08:
You can erase bonding info from CP with `_bleio.adapter.erase_bonding()`. You shouldn't need to do it every time though. The point of bonding is to allow for reconnection.

I don't think we should erase bonding info on upgrade because it may work just fine.

Please post the full code example and I'll try to reproduce it. What device are you pairing with?

rdoursenaud, 2021-09-08:
Thanks.

I guess I’m struggling because the documentation is lacking on the CircuitPython side of things regarding central mode.
Should I not scan again and attempt to connect directly if the device is already paired?

My main focus on this project is all things MIDI (5-Pin DIN/UART, USB, BLE and hopefully RTP).
I’m trying to get to grips with what can be done with the hardware.
What fails is re-connecting to a Yamaha MD-BT01 BLE/MIDI interface.

Here’s a more fleshed code snippet:
```py
# --- MIDI facilities ---
def test_midi_in(midi):
    print("Receiving MIDI")

    while True:
        data = midi.receive()
        if data:
            print(data)
            break

def test_midi_out(midi):
    print("Sending MIDI")
    print("Default output channel:", midi.out_channel + 1)

    note_hold = 0.85
    rest = note_hold / 5

    midi.send(ControlChange(64, 0))  # sustain CC
    midi.send(ControlChange(1, 0))  # modulation CC

    midi.send(NoteOn(48, 20))  # play note
    time.sleep(note_hold)  # hold note
    midi.send(NoteOff(48, 0))  # release note
    time.sleep(rest)  # rest

    midi.send(NoteOn(55, 40))
    time.sleep(note_hold)
    midi.send(NoteOff(55, 0))
    time.sleep(rest)

    midi.send(NoteOn(51, 60))
    time.sleep(note_hold)
    midi.send(NoteOff(51, 0))
    time.sleep(rest)

    midi.send(NoteOn(58, 80))
    time.sleep(note_hold)
    midi.send(NoteOff(58, 0))
    time.sleep(rest)

    midi.send(ControlChange(64, 32))
    midi.send(ControlChange(1, 127))
    midi.send(NoteOn(48, 20))  # play note
    time.sleep(note_hold)  # hold note
    midi.send(NoteOff(48, 0))  # release note
    time.sleep(rest)  # rest

    midi.send(NoteOn(55, 40))
    time.sleep(note_hold)
    midi.send(NoteOff(55, 0))
    time.sleep(rest)

    midi.send(NoteOn(51, 60))
    time.sleep(note_hold)
    midi.send(NoteOff(51, 0))
    time.sleep(rest)

    midi.send(NoteOn(50, 80))
    time.sleep(note_hold)
    midi.send(NoteOff(50, 0))
    time.sleep(rest)

def test_midi_io(midi):
    test_midi_in(midi)
    test_midi_out(midi)

def test_ble_midi_io(midi_service):
    midi = adafruit_midi.MIDI(midi_in=midi_service, midi_out=midi_service)
    test_midi_io(midi)

# --- nRF52840 BLE ---

print("Setting up Central BLE MIDI")
print("Using Adafruit BLE v{:s}".format(adafruit_ble.__version__))
BLE = adafruit_ble.BLERadio()

# Disconnect everything
if BLE.connected:
    for c in BLE.connections:
        c.disconnect()

# --- nRF52840 Central BLE MIDI ---

print("Setting up Central BLE MIDI")

print("Scanning BLE devices...")
advertisements = BLE.start_scan()

device_name = 'MD-BT01'
for advertisement in advertisements:
    print("Looking for {:s}".format(device_name))
    if advertisement.complete_name == device_name:
        print("{:s} found! Connecting...".format(device_name))
        BLE.stop_scan()
        BLE.connect(advertisement)
        if BLE.connected:
            print("{:s} connected!".format(device_name))
            connections = BLE.connections
        break

print("Pairing with {:s}...".format(device_name))
connection = connections[0]
connection.pair()
if connection.paired:
    print("{:s} paired!".format(device_name))
    if adafruit_ble_midi.MIDIService in connection:
        print("{:s} has MIDIService!".format(device_name))
        midi_service = connection[adafruit_ble_midi.MIDIService]

test_ble_midi_io(midi_service)
```

rdoursenaud, 2021-09-08:
Forgot to say that it fails on the `for advertisement in advertisements` with the same error message previously mentioned when the device has been successfully paired and interacted with on the previous run.

Also, clearing the bonding data with the advised `_bleio.adapter.erase_bonding()` before scanning again seems to behave as I was used to in CircuitPython 6.

tannewt, 2021-09-08:
If you are already connected to the device then there should be no need to disconnect. The new BLE workflow stuff in CircuitPython acts as a peripheral so it won't do scanning for you (that is a central thing.)

Can you replicate it with a shorter example? There is more BLE logic in 7.x that tries to load the bonding information when doing a scan in case the formerly bonded peer is advertising with a private random address (based on the exchanged key) instead of a public one.

rdoursenaud, 2021-09-08:
Good catch!

Not disconnecting behaves properly. I don’t know why I had that in place. Guess I copy/pasted sample code without thinking too much ;)

rdoursenaud, 2021-09-08:
Sorry for my previous comment, that’s not what’s going on.
It seems that running `_bleio.adapter.erase_bonding()` has definitively fixed the issue with my board since my original code now behaves as I expected.

Strange…

tannewt, 2021-09-08:
Does it work if you remove the erase_bonding line?

rdoursenaud, 2021-09-08:
Yes! That’s what I meant.

closed 2021-09-09 by dhalbert