adafruit/Adafruit_CircuitPython_BLE #136: HIDService(hid_descriptor=MY_HID_DESCRIPTOR)) causes NotImplementedError: Only one report id per Application collection supported

ThomasAtBBTF, 2021-09-18:
After https://github.com/adafruit/circuitpython/issues/4868 was closed I tried again a hid_descriptor with multiple report id's for one device.
Using the latest libraries it is clear, that "/lib/adafruit_ble/services/standard/hid.py", line 419 still has the catch for multiple report id's

```
  if len(reports) > 1:
    raise NotImplementedError(
      "Only one report id per Application collection supported"
  )
```

Also, it looks to me, that the following code, which adds ``ReportOut's`` and ``ReportIn's`` is not prepared to handle multiple report id's for one device. 

@dhalbert  Just commenting out the ``if`` statement does not help.

I think, that there are also problems elsewhere, because when using the standard_hid_descriptor with an added braille_hid_descriptor, else a standard keyboard from this descriptor is not working anymore.

I will upload a minimal sample file.

ThomasAtBBTF, 2023-04-18:
I hope to bring this issue up again with this message.
I checked in the latest version and see the code did not change and still only one OUT (from the host) per Application is supported.
In Discord I saw that there were questions about supporting Braille HID devices implemented in CP.
Braille HID is documented in hut1_4.pdf as device 0x41.
The host will send the braille dots to be displayed on the Braille Display using OUT reports.
"loosing" or "dropping" these reports is not an option as blind users will then not be able to read the correct texts from the host.
Apple is only supporting Braille HID for new displays in IOS and MACOS to connect to VoiceOver over BLE!
It would be a pity if a CP project could not be a Braille Display for VoiceOver...

ktnyt, 2024-10-19:
I think I got this to work. Let me open a PR!

closed 2024-11-13 by dhalbert