adafruit/Adafruit_CircuitPython_BLE #141: ValueError on BLERadio.start_scan() and ble_detailed_scan.py

bradjc, 2021-11-04:
I tried running `ble_detailed_scan.py` and get the following error:

```
scanning
Traceback (most recent call last):
  File "code.py", line 14, in <module>
  File "adafruit_ble/__init__.py", line 250, in start_scan
ValueError: Prefix buffer must be on the heap
```

I _think_ the error happens because the example scans for both services advertisements and any advertisement:

https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/4eca28e6407f511d5051fd608a238a9d8b40d783/examples/ble_detailed_scan.py#L18

which the causes the prefixes buffer to be empty:

https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/4eca28e6407f511d5051fd608a238a9d8b40d783/adafruit_ble/__init__.py#L244-L248

That seems to trigger the error here:

https://github.com/adafruit/circuitpython/blob/0dfb7a9a2da1444ebca4b8fff0b88795afede895/shared-bindings/_bleio/Adapter.c#L348-L355

I managed to get the example to work by either removing `Advertisement` from the scan_start() call, or commenting out the prefix check in `_bleio/Adapter.c`.

What I'm not sure about is the relevant code doesn't seem to have changed in the last 1.5 years. I'm using CircuitPython from the latest git commit on the micro:bit v2, so perhaps something weird is happening with my setup. But I thought I would open this issue just in case either something is wrong,.

kyril96, 2023-11-16:
I'm seeing this just now on CircuitPython 9.0.4 alpha on an ESP32-S3 QTPY -- with the non-detailed scan example, it just uses ble.start_scan() with no parameters.  Going back to CircuitPython 8 I no longer see that message. I'm not sure if this is a known issue or "you're doing it wrong" or if something got re-introduced.

tannewt, 2023-11-17:
> I'm seeing this just now on CircuitPython 9.0.4 alpha on an ESP32-S3 QTPY -- with the non-detailed scan example, it just uses ble.start_scan() with no parameters. Going back to CircuitPython 8 I no [...]

Please file an issue on CircuitPython if something breaks between versions. Its issues get more investigation than this library usually. https://github.com/adafruit/circuitpython/issues/

dhalbert, 2023-11-17:
I have a fix for this bug as a side effect of some other BLE debuggin I am doing, but I have seen more problems with ESP32-S3 BLE. I cannot get it to connect to anything as a central, and I tried many versions

@kyril96 Were you successfully connecting to a BLE peripheral, or just scanning advertisements?

kyril96, 2023-11-20:
@dhalbert With CP 8, I was able to get the advertisement and the "connection" object but getting the HIDService out of it choked. I think the best I was able to do is get a non crash-to-safemode error if I stopped scanning before I tried to get the connection from the advertisement, or with some side fumbling was able to get it to cough up that HIDService wasn't implemented.

@tannewt I popped the CP9 issue up there, we'll see if anyone gets to it. [CircuitPython issue 8631]( https://github.com/adafruit/circuitpython/issues/8631).

It seems like what I really want to do (homebrew the thing that the doomscroller or whatever feeds input to) is for now only doable with an nRF that doesn't come any smaller than an ItsyBitsy, or with Arduino or Espressif's own environment.

dhalbert, 2023-11-20:
It appears that service discovery, which happens after connection, never worked. I tested back to 7.3.3 and tested various 8.x.x versions as well.