adafruit/Adafruit_CircuitPython_BLE #164: Advertisement Names

MadGeometer, 2022-05-14:
Hello,

The advertisements returned by ble.start_scan() are missing short_name and complete_name. This happens in the ble_simpletest.py and ble_detailed_scan.py in the examples folder. Setting extended=true and increasing buffer_size doesn't help.

Any ideas?

Thanks,
Mike

CircuitPython Info: 
```
Adafruit CircuitPython 7.3.0-beta.2-6-gbf0e1fafa on 2022-05-02; Adafruit QT Py ESP32-S3 no psram with ESP32S3
Board ID:adafruit_qtpy_esp32s3_nopsram
```

Sample code:
```
from adafruit_ble import BLERadio

ble = BLERadio()

print("Beginning Scan")
foundAddresses = set()

for adv in ble.start_scan(extended=True, buffer_size=2048, timeout=5):
    addr = adv.address
    if addr not in foundAddresses:
        foundAddresses.add(addr)
        print(adv.short_name, adv.complete_name, adv.address, adv.connectable, adv.rssi, adv.appearance, adv.connectable, adv.scan_response)

ble.stop_scan()
print("Scan Complete")
```

Output:
```
Beginning Scan
None None <Address 57:9e:15:aa:10:4b> True -68 None True True
None None <Address 41:4a:36:7d:08:3c> True -72 None True True
None None <Address d6:0c:88:89:3b:55> False -65 None False False
Scan Complete
```

tannewt, 2022-05-16:
Are you sure the devices you are scanning advertises them? Using nRF connect on your phone is a good comparison.

MadGeometer, 2022-05-17:
Yes, the devices (one Macbook Pro, one iPad, and one iPhone) are indeed broadcasting their names. In fact, the names are displayed in Bluefruit Connect and nRF Connect, but not in the output from the sample code.

tannewt, 2022-05-17:
I'd recommend printing all of the advertisements. CircuitPython doesn't join the advertisement with scan response so you are at least missing half of the first two. IIRC you can print the raw advertisement with `bytes(adv)` too.