adafruit/Adafruit_CircuitPython_BLE #165: Unable to scan with

cygann, 2022-05-22:
Hello,

I'm having trouble finding success with the `ble_simpletest.py` example script ([here](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_simpletest.py)). The script runs, but no devices are ever found. I have verified that I can find and connect with my devices through nRF Connect. I'm running this script on an M1 Mac with MacOS 12.2.1. 

Wondering if anyone else has run across this and has a solution. Thanks!

ladyada, 2022-05-22:
cant repro - just ran that example on a super fresh m1 mac that is running latest everything and got adverts

```
 limorfried@Limors-MacBook-Air Desktop % pip3 freeze
Adafruit-Blinka==7.3.0
adafruit-blinka-bleio==2.2.1
adafruit-circuitpython-ble==8.2.4
adafruit-circuitpython-busdevice==5.1.8
adafruit-circuitpython-typing==1.7.0
Adafruit-PlatformDetect==3.22.1
Adafruit-PureIO==1.1.9
bleak==0.11.0
pyftdi==0.54.0
pyobjc-core==8.5
pyobjc-framework-Cocoa==8.5
pyobjc-framework-CoreBluetooth==8.5
pyobjc-framework-libdispatch==8.5
pyserial==3.5
pyusb==1.2.1
six @ file:///AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/python3/python3-110/six-1.15.0-py2.py3-none-any.whl
```

cygann, 2022-05-22:
Seeing that you had success with macOS 12.4 gave me a hunch to upgrade, and now the example works! Thanks!

I observed the same issue on another m1 MacBook running 12.2.1 as well. After a cursory search, it appears this likely a bug in macOS 12 prior to 12.3:
https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/33
https://github.com/hbldh/bleak/issues/720#issuecomment-1009198158

closed 2022-05-22 by cygann