adafruit/Adafruit_CircuitPython_BLE #171: Add Telephony Device Page (0x0B) HID_DESCRIPTOR

ThomasAtBBTF, 2022-08-11:
It would be very helpful if someone could add a sample HID_DESCRIPTOR for the Telephony Device Page (0x0B)
So we could make a ItsyBitsy Project to answer or decline incoming calls on an IPhone.
(And play with other Usages)

ThomasAtBBTF, 2022-08-11:
I think the "only" thing we need is a small collection of bytes that can be added to the standard HID Descriptor for advertising the peripheral capabilities.

aguaviva, 2024-12-03:
Supporting a telephony would be very useful as it would allow creating a "panic button" for old people so it calls their sons. I'd also make a nice kit with a nice red button.

Nobody has done this before and you would be the first one.