adafruit/Adafruit_CircuitPython_BLE #177: BLEConnection.disconnect doesn't work on

xorbit, 2022-11-27:


xorbit, 2022-11-27:
Sorry, fat fingers.

I have the following piece of code to disconnect my macropad from all devices if a button is held on boot:
```python
ble = adafruit_ble.BLERadio()
if not ble.connected:
  print ("Advertising")
  ble.start_advertising(advertisement, scan_response)
else:
  print ("Already connected")
  print (ble.connections)
  if buttons[4].value == 0:
    print ("Requesting disconnect")
    for connection in ble.connections:
      connection.disconnect()
```

However, it doesn't seem to do anything.  The previously connected device stays connected.
Am I doing this wrong or is this a bug?

tannewt, 2022-11-28:
Do you have the BLE workflow active? It should work if not and this sounds like a bug.

xorbit, 2022-11-28:
Hi Scott, I'm not entirely certain what you mean by "do you have the BLE workflow active".  The BLE code is working, the macropad can communicate with devices using code that was built from the Adafruit BLE HID keyboard example.  The only problem I have now is that I can't seem to find a way to disconnect the macropad from the device it's currently connected to so it can be connected to something else.  I was trying to do that with the piece of code above (disconnect all connections if a button is held when turning the macropad on).