adafruit/Adafruit_CircuitPython_BLE #178: NotImplementedError:  On UARTService()

Minus-hates-school, 2022-12-23:
ESP32 Pro S3 using CircuitPython

Error Message: 

`Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "/lib/adafruit_ble/services/nordic.py", line 55, in __init__
  File "/lib/adafruit_ble/services/__init__.py", line 49, in __init__
NotImplementedError: `

Version / modle: 

`Adafruit CircuitPython 7.3.1 on 2022-06-22; ProS3 with ESP32S3`

Code(color picker example):

```
import board
import neopixel

from adafruit_bluefruit_connect.packet import Packet
from adafruit_bluefruit_connect.color_packet import ColorPacket

from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

ble = BLERadio()
uart_service = UARTService()
advertisement = ProvideServicesAdvertisement(uart_service)

pixels = neopixel.NeoPixel(board.NEOPIXEL, 10, brightness=0.1)

while True:
    # Advertise when not connected.
    ble.start_advertising(advertisement)
    while not ble.connected:
        pass
    ble.stop_advertising()

    while ble.connected:
        if uart_service.in_waiting:
            packet = Packet.from_stream(uart_service)
            if isinstance(packet, ColorPacket):
                print(packet.color)
                pixels.fill(packet.color)
```

Neradoc, 2022-12-23:
Hi, services are not implemented on ESP32 chips in Circuitpython.
See: https://github.com/adafruit/circuitpython/issues/5926#issuecomment-1074414184
And: https://learn.adafruit.com/welcome-to-circuitpython/frequently-asked-questions#faq-3129409

closed 2022-12-24 by Minus-hates-school

shanecromer24, 2024-03-20:
I am reading that there is support in Circuit Python 9.0 but there is code that need to be written for the implementation.