adafruit/Adafruit_CircuitPython_BLE #179: How to protect BLE connection?

paluigi, 2022-12-27:
Hello,

I am new to CircuitPython and I am trying to create a protected BLE connection.  Between documentation and examples I cannot understand how to create a protected BLE connection (MITM and possibly a PIN/passkey validation to accept connection requests).

Desired status:

1. Advertise a BLE connection with MITM protection
2. Validate BLE connection requests against a PIN/passkey programmed on the device

Device: [MDBT50Q-RX](https://circuitpython.org/board/raytac_mdbt50q-rx/) with nRF52840 .
CircuitPython version: 7.3.3

I see in the library there are suitable [Attributes](https://docs.circuitpython.org/projects/ble/en/latest/attributes.html) for the MITM part (for instance: Attribute.LESC_ENCRYPT_WITH_MITM), but I cannot figure out  how to use them in the process of creating and advertising services.

For PIN authentication, instead, I can't find anything. Just as a reference, in the Arduino world I would follow this [model](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Peripheral/pairing_passkey).

Here the current code I am using, which successfully creates a non-secure connection (I am using the Bluefruit Connect app to test):

```
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

ble = BLERadio()
uart = UARTService()

advertisement = ProvideServicesAdvertisement(uart)
ble.start_advertising(advertisement)

while not ble.connected:
    pass

while True:
    if ble.connected:
        s = uart.readline()
        if s:
            try:
                print(str(s.decode()))
            except Exception as e:
                print(e)
                uart.write(f"{e} - ERROR\n".encode("utf-8"))
            uart.write(f"{str(s.decode())} - DONE\n".encode("utf-8"))

```
Any help would be appreciated! Thanks.

dhalbert, 2022-12-27:
Sorry -- passkey/PIN validation is not supported right now, and also I don't think there is a way to set the MITM protection either. "Just works" pairing/bonding is what we provide for now.

paluigi, 2022-12-27:
Thanks for the quick reply! Then for this project we'll have to stay on the Arduino side.
It's a pity, as I felt in love with the idea to use Python also on micros! I'll keep an eye on future updates.

closed 2023-01-06 by tekktrik