adafruit/Adafruit_CircuitPython_BLE #181: NotImplementedError: in FeatherS3 - ESP32-S3 adafruit

WedahithaYapa, 2023-01-26:
Hi , I am trying to upload a BLE HID code to FeatherS3 - ESP32-S3 adafruit and gets a NotImplementedError. I have no idea how to proceed forward. Please help me on this. THis code works perfect on a nrf52 series chip.
Here is my Code

////////////////////////////////////////////////////////////

import time
import board
import digitalio
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS

import adafruit_ble
from adafruit_ble.advertising import Advertisement
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.standard.hid import HIDService
from adafruit_ble.services.standard.device_info import DeviceInfoService

 
hid = HIDService()
device_info = DeviceInfoService(
    software_revision=adafruit_ble.__version__, manufacturer="Adafruit Industries"
)
advertisement = ProvideServicesAdvertisement(hid)
advertisement.appearance = 961
scan_response = Advertisement()

ble = adafruit_ble.BLERadio()
if ble.connected:
    for c in ble.connections:
        c.disconnect()

print("advertising")
ble.start_advertising(advertisement, scan_response)

k = Keyboard(hid.devices)
kl = KeyboardLayoutUS(k)
while True:
    while not ble.connected:
        pass
    while ble.connected:
        kl.write("x")
        time.sleep(1)
    ble.start_advertising(advertisement)

////////////////////////////////////////////////////////////

the error I am getting is attached

[image]

>

Neradoc, 2023-01-26:
Services are not implemented in Circuitpython on ESP32, see:
https://github.com/adafruit/circuitpython/issues/5926
https://learn.adafruit.com/welcome-to-circuitpython/frequently-asked-questions#faq-3129409

You can use an NRF52 board to make an HID device, or look into solutions in Arduino.

WedahithaYapa, 2023-01-26:
Thank you. This is very Helpful

closed 2023-01-26 by WedahithaYapa