adafruit/Adafruit_CircuitPython_BLE #186: Pyportal Titano BLE advertisement length error

UnfinishedStuff, 2023-04-29:
Hi everyone,

This issue was originally posted on the Adafruit Forums: https://forums.adafruit.com/viewtopic.php?p=970491#p970491

I've been trying to use BLE with CircuitPython, and I've hit a snag with the PyPortal Titano.  I have:

- Updated CircuitPython to version 8.0.5
- Upgraded the libraries on the device to those from the 8.x bundle
- Updated the Airlift firmware to NINA_W102-1.7.4 using Adafruit's guide

<details>

```python

# SPDX-FileCopyrightText: 2020 Dan Halbert, written for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense

# pylint: disable=unused-import
import board
import busio
from digitalio import DigitalInOut
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService
from adafruit_esp32spi import adafruit_esp32spi
from adafruit_airlift.esp32 import ESP32

# If you are using a Metro M4 Airlift Lite, PyPortal,
# or MatrixPortal, you can use the default pin settings.
# Leave this DEFAULT line uncommented.
# If you are using a board with pre-defined ESP32 Pins:
esp32 = ESP32()

# If you are using a Metro M7 **OR**
# if you are using CircuitPython 6.0.0 or earlier,
# on PyPortal and PyPortal Titano only, use the pin settings
# below. Comment out the DEFAULT line above and uncomment
# the line below. For CircuitPython 6.1.0, the pin names
# have changed for these boards, and the DEFAULT line
# above is correct.
# esp32 = ESP32(tx=board.TX, rx=board.RX)

# If you are using an AirLift FeatherWing or AirLift Bitsy Add-On,
# use the pin settings below. Comment out the DEFAULT line above
# and uncomment the lines below.
# If you are using an AirLift Breakout, check that these
# choices match the wiring to your microcontroller board,
# or change them as appropriate.
# esp32 = ESP32(
#     reset=board.D12,
#     gpio0=board.D10,
#     busy=board.D11,
#     chip_select=board.D13,
#     tx=board.TX,
#     rx=board.RX,
# )

# If you are using an AirLift Shield,
# use the pin settings below. Comment out the DEFAULT line above
# and uncomment the lines below.
# esp32 = ESP32(
#     reset=board.D5,
#     gpio0=board.D6,
#     busy=board.D7,
#     chip_select=board.D10,
#     tx=board.TX,
#     rx=board.RX,
# )

adapter = esp32.start_bluetooth()

ble = BLERadio(adapter)
uart = UARTService()
advertisement = ProvideServicesAdvertisement(uart)

while True:
    ble.start_advertising(advertisement)
    print("waiting to connect")
    while not ble.connected:
        pass
    print("connected: trying to read input")
    while ble.connected:
        # Returns b'' if nothing was read.
        one_byte = uart.read(1)
        if one_byte:
            print(one_byte)
            uart.write(one_byte)

```
<summary>
When I try to use this code 
</summary></details>

[From this Adafruit Guide](https://learn.adafruit.com/adafruit-pyportal/circuitpython-ble)

I get this error from Mu:

```python
0;🐍main.py | 8.0.5\]0;🐍main.py | 8.0.5\Traceback (most recent call last):
  File "main.py", line 64, in <module>
  File "adafruit_ble/__init__.py", line 215, in start_advertising
ValueError: Data too large for advertisement packet
]0;🐍215@adafruit_ble/__init__.py ValueError | 8.0.5\
Code done running.
```

I've also done all of this using a Feather M4 Express with the standalone Airlift breakout (ADA4201) and it works just fine, so this seems to be something specific to the Titano or built-in airlifts.  The code also works fine when run on a Feather nRF52840 and a Seeed Xiao nRF52840.

Mikeysklar on the forum asked me to print `advertisement` after the code runs:

```python

uart = UARTService()
advertisement = ProvideServicesAdvertisement(uart)
```

On the Titano this gives:
`<ProvideServicesAdvertisement flags=<AdvertisingFlags general_discovery le_only > services=<BoundServiceList: UUID('6e400001-b5a3-f393-e0a9-e50e24dcca9e')> >`

On the Feather M4 Express with Airlift breakout this gives:
`<ProvideServicesAdvertisement services=<BoundServiceList: UUID('6e400001-b5a3-f393-e0a9-e50e24dcca9e')> flags=<AdvertisingFlags general_discovery le_only > >`

The BLE code is a little beyond me, so I can't quite see what's going wrong here.  Any help would be very much appreciated.

dhalbert, 2023-10-28:
Could you please update CircuitPython to 8.2.7 and re-test, and could you also confirm that the firmware is at 1.7.4? You can confirm the firmware version using this Internet Connect test program: https://learn.adafruit.com/adafruit-pyportal/internet-connect

UnfinishedStuff, 2023-10-28:
Thanks for the suggestions @dhalbert.  No, I didn't get it sorted so I abandoned the project, but I would love to get it working.

I've updated CircuitPython as well as all libraries and I still get the same error:

```python
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
]0;🐍main.py | 8.2.7\Traceback (most recent call last):
  File "main.py", line 64, in <module>
  File "adafruit_ble/__init__.py", line 216, in start_advertising
ValueError: Data too large for advertisement packet
]0;🐍216@adafruit_ble/__init__.py ValueError | 8.2.7\
Code done running.

```

>could you also confirm that the firmware is at 1.7.4?

Yes, the output states `Firmware vers. bytearray(b'1.7.4\x00')`, so that seems fine.

The `advertisement` continues to look as it did before:

```
<ProvideServicesAdvertisement flags=<AdvertisingFlags general_discovery le_only > services=<BoundServiceList: 
UUID('6e400001-b5a3-f393-e0a9-e50e24dcca9e')>
```

Any further suggestions would be much appreciated!