adafruit/Adafruit_CircuitPython_BLE #188: Unable to scan for extended advertising

Nilon123456789, 2023-06-21:
I'm trying to listen for extended advertising from an nrf52833. I tried the [ble_detailed_scan.py](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_detailed_scan.py) script, by adding the `extended=True` flag. I know that my bluetooth adapter is compatible since I can detect my device in the [BluetoothLEExplorer](https://github.com/microsoft/BluetoothLEExplorer) from microsoft.

tannewt, 2023-06-21:
What platform are you running this code on?

Nilon123456789, 2023-06-21:
Windows 10

tannewt, 2023-06-21:
I don't think this is a bug with this library. I suspect it is an issue with https://github.com/hbldh/bleak which we use underneath blinka bleio.