adafruit/Adafruit_CircuitPython_BLE #191: Failed to connect: internal error

CptanPanic, 2023-11-16:
This is on a ESP32-S3 Reverse TFT, connecting to a BLE device. I get "Failed to connect: internal error"

Output:
```
code.py output:
Scanning for BLE devices...
found target device!
address: <Address 6c:79:b8:bb:b1:ed>
connectable: True
flags: <AdvertisingFlags general_discovery le_only >
matches: <bound_method>
rssi: -69
scan_response: True
tx_power: None
complete_name: None
get_prefix_bytes: <bound_method>
data_dict: {2: b'\xf0\xff', 1: <AdvertisingFlags object at 0x3de0dd20>}
mutable: False
match_prefixes: ()
_prefix_bytes: b''
short_name: None
Stopped Scan
Error connecting to robot: Failed to connect: internal error

Code done running.
```

```
import time
import adafruit_ble
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

target_address = '6c:79:b8:bb:b1:ed'
target_address_bytes = bytes(reversed([int(x, 16) for x in target_address.split(":")]))

print("Scanning for BLE devices...")
ble = adafruit_ble.BLERadio()
devices = ble.start_scan(timeout=1, extended=True, interval=.2)

for device in devices:
    if device.address.address_bytes == target_address_bytes:
        print("found target device!")
        for attr in ['address', 'connectable', 'flags', 'matches', 'rssi', 'scan_response', 'tx_power', 'complete_name', 'get_prefix_bytes', 'data_dict', 'mutable', 'match_prefixes', '_prefix_bytes', 'short_name']:
            print(f"{attr}: {getattr(device, attr)}")
        
        try:
            ble.stop_scan()
            print("Stopped Scan")

            robot_connection = ble.connect(device)

            if robot_connection and robot_connection.connected:
                print("Connected to robot")
                uart_service = robot_connection[UARTService]
                uart_service.write(0xfff2)
                print("Sent message to robot")
        except Exception as e:
            print("Error connecting to robot:", e)
        break
```

henrygab, 2025-03-03:
Adafruit's ESP32-S3 ReverseTFT board comes with 4MB flash.  IIRC, for CircuitPython9, `_ble` wasn't expected to be supported without at least 8MB flash.   Did that change?  How are you getting BLE in CircuitPython on a 4MB flash ESP32-S3 device?

Neradoc, 2025-03-03:
> Adafruit's ESP32-S3 ReverseTFT board comes with 4MB flash. IIRC, for CircuitPython9, `_ble` wasn't expected to be supported without at least 8MB flash. Did that change? How are you getting BLE in CircuitPython on a 4MB flash ESP32-S3 device?

In November 2023, that would have been on Circuitpython 8, which supported BLE but in client mode only, it did not support starting a service on the board. FYI you can now jump through some hoops to make a build with BLE on 4MB boards as I mentioned in this PR: https://github.com/adafruit/circuitpython/pull/10089
This issue is likely quite obsolete by now.

dhalbert, 2025-03-03:
It needs re-test, on a board that will support it now. The actual board is not import, as long as it's an ESP32-S3 or even any Espressif board with BLE enabled.