adafruit/Adafruit_CircuitPython_BLE #194: How to utilize ServiceData advertising field? (Improv WiFi)

konistehrad, 2024-04-16:
I’m currently implementing [Improv WiFi](https://www.improv-wifi.com/) in CircuitPython for release as a library, but I’m having trouble hitting one part of the specification, namely the [service data format](https://www.improv-wifi.com/ble/). Namely, I don’t see any example code on how to attach it to an Advertisement, or how to modify its UUID to the requisite value of  4677.  Any assistance you could provide would be much appreciated. Thanks!

dhalbert, 2024-04-16:
`ServiceData` is a class in `advertising/standard.py`. You pass its constructor a `Service`, which includes a UUID. There is an an example of using `ServiceData` in the Eddystone beacon library: https://github.com/adafruit/Adafruit_CircuitPython_BLE_Eddystone/blob/main/adafruit_ble_eddystone/__init__.py

konistehrad, 2024-04-16:
Right, I did see that. The Eddystone tip is a good one I'm going to eval that now. In the meantime, are the type annotations for `ServiceData`'s constructor incorrect? They claim to take a Characteristic, should that be a Service?

https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/adafruit_ble/advertising/standard.py#L356

dhalbert, 2024-04-16:
Yup, that's definitely an annotation bug -- thanks for pointing that out. I'll change the title and submit a PR for that.

closed 2024-04-16 by tannewt

konistehrad, 2024-04-16:
Awesome thanks for being so quick on this. I'm not sure if this is the best place to continue the conversation, but what's the correct way to consume `StreamIn` and `StreamOut`? In particular, there's an RPC Command characteristic that must receive multiple packets to form itself, and while I've found some templates to do this (in for example adafruit_ble_apple_notification_center) nothing concrete has shown its head. Is there some kind of template code to use Generators to monitor characteristics, or is that the best to reference?

tannewt, 2024-04-24:
> Awesome thanks for being so quick on this. I'm not sure if this is the best place to continue the conversation, but what's the correct way to consume `StreamIn` and `StreamOut`? In particular, [...]

[Discord's #circuitpython-dev](https://adafru.it/discord) may be a better place. If you are using Adafruit hardware then the [technical support forum](https://forums.adafruit.com/viewforum.php?f=60). The UART example is probably the best use reference: https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_uart_echo_client.py

PR #195 (2024-04-16): Correct type annotations in advertising/standard.py
- Fixes #194